Ryzee119/ogx360 #2: Increase reliability with non official wireless adapters

hugo19941994, 2019-04-20:
Hi

I haven't seen anyone report any similar issues, but with my chinese knock-off Xbox 360 wireless adapter I had to manually press reset every time I powered on the Xbox for the ogx360 to sync.

Compiling with increased delays (2000) in main.cpp has solved this issue for me. A much lower value would have probably solved it as well, but as it worked I left it like that.

https://github.com/Ryzee119/ogx360/blob/a7b2420ace5b00d294adeca6a61302b18189174f/Firmware/ogx360_32u4/ogx360_32u4/main.cpp#L155-L159

I've installed the ogx360 internally so it's now a bit of a chore to try other values. However I'm waiting for some micro usb connectors to make the install a bit neater. Whenever those arrive I'll try some other values to see how low I can get them while still making the receiver connect reliably.

Ryzee119, 2019-04-22:
Thanks Hugo, ill investigate and see what I can do

hugo19941994, 2019-05-04:
Hello again. I've managed to make it work by increasing the first delay (line 155 in main.cpp) to 400ms. The other delays at 20ms were fine.

I actually managed to get it working with a 320ms delay, but it was unable to connect once (out of 10 tests or so), so I've settled on 400ms.

TEK-Nemesis, 2019-05-28:
My buddy had the same problem with his clone receiver.  I fixed it by reprogramming the OGX360 master and slaves with setting the delay on line 155 (in main.cpp) to 400ms.  (Thanks, Hugo19941994!)

Ryzee119, 2019-05-28:
I was unable to replicate this issue with my clone receiver but seeing as its happened to a few people ill commit a new change shortly.

Ryzee119, 2019-06-01:
Merged into master - commit# 2846542

closed 2019-06-01 by Ryzee119