Ryzee119/ogx360 #3: Rumble makes controller disconnect

TEK-Nemesis, 2019-05-22:
Area 51 with rumble on makes the controller disconnect on every rumble.  Turning rumble off solves the issue.  

Only one wireless controller was connected.
Rev 1.0 board.  128MB RAM.  X3CE modchip using X2 5035.  
Genuine Xbox 360 controller with genuine receiver.

Ryzee119, 2019-05-22:
Thanks foe the feedback
Is it only that game? Have you tried new batteries in the wireless controller?

TEK-Nemesis, 2019-05-22:
I only tried that game last night.

One battery (Panasonic alkaline) is at 1.52v; the other is at 1.14v (for a total of 2.66v).  Will replace both with Duracell alkaline batteries at 1.61v each (for a total of 3.22v).  

(My gut feeling is that this will work.  I'll report back once tested.)

TEK-Nemesis, 2019-05-22:
Area 51 tests perfectly with rumble.  Thanks for your quick assessment!

closed 2019-05-22 by TEK-Nemesis

Ryzee119, 2019-05-23:
No problem, I will look into mayble disabling rumble on low battery levels automatically to prevent this happening and extend controller life