Ryzee119/ogx360 #5: Grey 8Bitdo USB Adapter Support

SlowCatalyst, 2019-06-08:
Is the grey 8bitdo USB adapter supported as well?

I remember talking with you recently about the n360, in that there was a possibility of 8bitdo USB adapters being implemented to the firmware, but the grey model was a bit up in the air.

I just received my ogx360 [this thing is fkn beautiful by the way- amazing job]- hooked up the grey 8bitdo USB adapter, and NES30 Pro connected to it fine, however there is no input detected.

Am I missing a simple step, or does the grey USB infact have a different device signature- or something of that nature- from the orange variant USB adapter?

Ryzee119, 2019-06-08:
Only the orange adaptor has been tested.
If you can provide a usb descriptor dump from a windows PC and I can have a look at it to see how it differs.
I dont own a grey one to test

Evelsnake, 2019-08-03:
[DescriptorDump_Controller.txt](https://github.com/Ryzee119/ogx360/files/3463269/DescriptorDump_Controller.txt)

Information for device Controller (VID=0x054C PID=0x0CDA):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0002
Current configuration value: 0x01
Number of open pipes: 2

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x054C	idVendor
0x0CDA	idProduct
0x0100	bcdDevice
0x01	iManufacturer   "Sony Interactive Entertainment"
0x02	iProduct   "Controller"
0x00	iSerialNumber
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x003B	wTotalLength   (59 bytes)
0x02	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
0x32	bMaxPower      (100 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0031	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0110	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0032	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x82	bEndpointAddress  (IN endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Sony Interactive Entertainment"
0x02   0x0409  "Controller"

------------------------------

Connection path for device: 
Intel(R) 6 Series/C200 Series Chipset Family USB Enhanced Host Controller - 1C26
Root Hub
Generic USB Hub
Controller (VID=0x054C PID=0x0CDA) Port: 5

Running on: Windows 10 or greater

Brought to you by TDD v2.11.0, Mar 26 2018, 09:54:50

Ryzee119, 2019-08-03:
Thanks mate, appreciate it.
It enumerates as a PS Classic controller; this is currently not implemented and it's completely different from the Orange 8bitdo adaptor (which is supported). So won't work yet.

Doesn't look difficult to write a driver for it, but I won't be able to commit to anything yet. I'm also at the complete limits of Flash memory available

Ryzee119, 2019-08-11:
ok as a workaround in the interim, if anyone has the grey 8btido adaptor, I have a theory that the firmware from the orange unit can be written to it. 
You can download the firmware for the orange unit from here:
https://support.8bitdo.com/

It is completely reversible.

I have tested converting my orange adaptor to the grey one myself. But no this way around. I assume it works the same.

CroCop86, 2019-08-11:
I have flashed the orange Firmware to the grey unit, but ogx360 won't recognize it

Ryzee119, 2019-08-11:
Thanks for testing @CroCop86. Well there goes that theory 😄 . I will be getting the grey adaptor soon so can do some testing from it too

Ryzee119, 2019-08-24:
So I got my grey adaptor and if I manually update the firmware with the firmware for the Orange adaptor it works for me https://download.8bitdo.com/Firmware/Receiver/usb_adapter/8BitDo_USB_RR_Firmware_V1.27.zip

[image]

Ryzee119, 2019-08-24:
If it doesn't work try updating to the latest firmware. I have improved some timings which may make a difference.

Kilerz28, 2020-01-18:
Hey. I have done what you said here @Ryzee119 and it works for me. I have the gray adapter and it pairs with my xbox one controller. Thanks.

Ryzee119, 2020-03-15:
Closed, as it is simple to write the orange/brown unit fimmware to the grey device so that it works.

closed 2020-03-15 by Ryzee119