Ryzee119/ogx360 #6: Xbox Wireless Adapter for Windows 10 Support

johndoenulc, 2019-06-12:
Any chance of support for the newer xbox wireless adapters? the ones that come with the xbox one controllers

https://www.xbox.com/en-GB/xbox-one/accessories/adapters/wireless-adapter-windows

Ryzee119, 2019-06-14:
Unless reverse engineering efforts are completed for the protocol I cannot write drivers for this. Unfortunately I dont really have time to look into this myself. I will leave issue open incase someone can chime in with info.
Current solution is to use the Orange 8bitdo adaptor and connect to it via bluetooth (limited to 1 controller per 8bitdo adaptor)

xandayn, 2020-03-26:
Hi, I'm not sure if this will be useful, but there is [a linux driver](https://github.com/medusalix/xow) that is being worked on for the xbox one wireless dongle (I don't own one myself so I can't test it personally). Not sure how helpful (if any) this information is and I don't have the knowledge to be able to implement this myself, but I thought I should provide it for anyone who does.

irlandez, 2020-05-02:
> Current solution is to use the Orange 8bitdo adaptor

hi. if you take the gray adapter will not work?

Ryzee119, 2020-05-02:
Grey adaptor will work if you update the firmware with the orange firmware. You can get it from the 8bditdo site

anthrox, 2020-05-26:
Hi @Ryzee119 found this project https://github.com/medusalix/xow

GoTeamScotch, 2020-11-28:
Wanted to chime in and say that I would donate towards development for X1 wireless adapters. I would rather give you, Ryzee, a hundred bucks over buying 4 8bitdo's + a USB hub, which adds up to about the same amount. Maybe there's others out there who would be willing to chip in too to help make it worth your time.

Ryzee119, 2020-12-08:
Ive looked into this and the 32u4 simply isn't powerful enough to support this. I have a port on the Teensy4.1 which opens up this possibility. It will likely be something I can look at in more detail there.
https://github.com/Ryzee119/ogx360_t4

closed 2020-12-08 by Ryzee119