Ryzee119/ogx360 #9: 3 pad RED LED 67-1556-1-ND

tai1976, 2019-08-01:
Had some boards made up from OSHPark and used your Digikey quick cart and everything went and works well. However, the 3 pad RED LED 67-1556-1-ND has really thrown me for a loop.
I believe the center pad is for heat dissipation which can increase brightness at lower voltages and the overall life of the LED but your PCB layout only has 2 pads for the anode and cathode and I was concerned the center pad could possibly short both pads.
So my question is this. Why use a 3 pad LED with only 2 pads on the PCB?
Was 67-1556-1-ND originally a 2 pad LED? Should I solder the LED a bit off center so that the center pad dissipates through the anode or cathode or am I just missing something?

Ryzee119, 2019-08-02:
That's interesting, 67-1556-1-ND wasn't a 3-pad LED in my prototyping from memory.
It's unclear exactly what that 3rd pad is, just mechanical support I guess, and some heat dissipation but that wont be a concern for this.

I'd say that centre pad can remain unconnected. I will update that part to a more standard 0805 LED. It wont impact the operation of the board in anway, purely a power indication LED (But the arduinos have this anyway)

Thank you for highlighting the issue.

tai1976, 2019-08-02:
Thank you for getting back. That is what I thought. I just used a standard 0603 LED I already had on hand and swapped R5 for a 220 Ohm resistor to increase the brightness. 
Thank you so much for making and publishing this project. This is truly a must have/make for any OG Xbox fan.

Ryzee119, 2019-08-03:
Closed by PR #9fa276d

closed 2019-08-03 by Ryzee119