Ryzee119/ogx360 #10: generic wireless adapter not syncing

loggio, 2019-08-02:
Hi, 

I have obtained an OGX360 and can't thank you enough for all your hard work. It's awesome! 

I'm having issues however... I have a generic wireless adapter off eBay.
It will allow me to sync 3 of my xbox 360 controllers to the OGX360, but the other 4 controllers i have do not sync. Funny enough all of the controllers that i cannot get to sync are all white version controllers (could be just coincidence). Maybe they're older models??

I suspect that your driver support is not working for certain versions of the 360 controller. A handshake miss-match so to speak. 

When the ogx360 is plugged in to the control ports of the XBOX, i push the sync buttons and the controller will acknowledge the dongle as it stops the sync process and begins the quick pulse handshake, the master controller of the OGX360 will briefly flash red and then disappear as if it cannot pair with the controller.... This loop continues over and over until the controller battery is pulled.

If i plug this receiver into my windows 10 computer and install the genuine microsoft driver for the dongle, i can sync the controllers just fine to my PC. 

All of my controllers work perfectly and sync with the PC and the XBOX 360. But only 4 out f my 7 controllers will sync successfully with the OGX360, the other 3 will not.

Is there any information i can supply you that will help with diagnosing this problem ? 
hopefully a firmware update will fix this.

Ryzee119, 2019-08-02:
Hey mate,
Thanks for the details response. 
Its possible there is some differences that Im not aware of.

Out of curiosity, the controllers that don't appear to sync, have you checked if the buttons etc. actually work still? 
Its possible its just the LED thats not setting to a quadrant correctly. 

If the controllers don't work at all we can look into it a bit further

loggio, 2019-08-02:
They don't work at all. The controllers will recognise the dongle but won't sync at all.

Ryzee119, 2019-08-02:
My understanding was that all the syncing and negotiations was handled by the receiver. The driver doesn't do any of that so this is an interesting one.

I'll have to do some digging this weekend to see how the windows driver handshakes and if im doing anything differently.

loggio, 2019-08-02:
Yes it's weird. They're all genuine controllers. The only thing different is the Chinese dongle. Though like I said, the same dongle syncs the controllers fine on Windows pc.

And the controllers sync fine on my xbox 360. Just not the Ogx360.

Let me know if you need me to do any further testing.

Ryzee119, 2019-08-06:
So i've dumped the USB traffic going to the wireless receiver in Windows on connection of the Xbox 360 controller.

The following packets are sent to the respective endpoint (1,3,5 or 7 depending on the controller)
00 00 08 40 00 00 00 00 00 00 00 00 //Turn LED off

00 00 08 42 00 00 00 00 00 00 00 00 //Set LED to blink. 0x42 is for quadrant 1

00 00 02 80 00 00 00 00 00 00 00 00 //Unknown?

00 00 00 40 00 00 00 00 00 00 00 00 //Update battery level I think

00 00 08 46 00 00 00 00 00 00 00 00 //Set LED to solid on for quadrant 1

It also polls every 2.5 seconds, but I think this is just to update the battery level or some kind of status to the windows driver.

I have implemented this init sequence exactly as per the windows driver

[ogx360_issue10.zip](https://github.com/Ryzee119/ogx360/files/3471615/ogx360_issue10.zip)

If you are able to test this firmware and report back that would be awesome!

Programming instructions are here

https://github.com/Ryzee119/ogx360/tree/master/Firmware#programming

loggio, 2019-08-06:
just tested and still no difference. Cannot sync

loggio, 2019-08-06:
i do have a microsoft branded dongle coming in the mail... When i receive that i will test and report back. Will be interesting if it is the dongle. 

But as i said, the dongle works fine on the PC syncing with all my controllers using the official microsoft drivers. So not sure what could be causing this. 

The controllers seem to sync with the dongle, but they won't latch on to a port. you can see that one micro controller will light up for briefly and then disappear

Ryzee119, 2019-08-06:
Thanks for your prompt testing
That will be very interesting to see if the genuine one works
I'm also curious to know if it works in Linux as my driver is loosely based on the xpad driver in Linux.

loggio, 2019-08-06:
Ok,  I can test on Ubuntu. Hopefully it's just plug and play. I will do this tomorrow.

loggio, 2019-08-06:
So, just to summaries... 

 My steps 

1. Turn on xbox. Wait for boot.
2. Attach dongle 
3. Attach master to Port 1 of xbox
4. Turn on controller, press sync.
5. Press sync on dongle.

Results.

After pressing sync on both controller and dongle, the controller will very quickly sync with the dongle as the controller stops the sync cycle and starts to flash.

I see the master controller of the Ogx360 flash red briefly, but won't latch on. The controller will just sit in a pairing (flashing) state until switched off. Slave controllers act the same.

The handshake between the controller, dongle and Port assignments seems to not want to cooperate. 

It sounds very much like I'm using different versions of controllers and some of which are incompatible with the drivers.

I just can't see why some controllers work fine, and others won't.

Ryzee119, 2019-08-06:
Thanks for the summary, I'll have a think about it and see if I can come up with anything in the meantime too

loggio, 2019-08-07:
So, i've layed out all my controllers. the three white controllers stacked are the ones that do not work. 

Interestingly enough, they are a different model number to the rest - model MSK-WC01

The black controllers and the silver controllers are model number MSK-1403 
The single white controller is model number  MSK-XBOX360WLSCTLR

All work on PC and seem to sync with the dongle. 
only model number MSK-WC01 will not work with your OGX360.

[image]
[image]
[image]
[image]
[image]

I have yet to try with Ubuntu. Will do this shortly.

loggio, 2019-08-07:
Ok, the controller syncs and works fine on Ubuntu. 
See videos below.

[videos.zip](https://github.com/Ryzee119/ogx360/files/3475838/videos.zip)

Ryzee119, 2019-08-07:
legend mate, thanks once again for the detailed response. I'll review/test and see what I come up with!
Unfortunately all my controllers are MSK-1403, so there's something interesting going on with that model I guess

loggio, 2019-08-07:
One thing I noticed is that it took a little while to assign a quadrant for the controller on Ubuntu... Maybe this model takes a little longer to sync up? 

Might be a timing thing perhaps?

Ryzee119, 2019-08-09:
[ogx360_issue10_1.zip](https://github.com/Ryzee119/ogx360/files/3485933/ogx360_issue10_1.zip)
Ok, so I tweaked some timing to match the Windows driver more closely, and also changed all my packet sizes to 12 bytes.

Ryzee119, 2019-08-09:
You probably only need to update the MASTER device for this, so no need to update all modules

loggio, 2019-08-09:
Hmm it's still not working.

However, it seems to latch for a second or two longer and more frequently ... Enough to register a quadrant but won't lock on.

Ryzee119, 2019-08-09:
wow those are some picky controllers. I think I'm replicating the Windows driver exactly. Last thing to see is if the genuine receiver makes a difference

loggio, 2019-08-09:
It's definitely trying a lot harder

Ryzee119, 2019-08-09:
The controller's not turning off right? Just disconnecting?

loggio, 2019-08-09:
Hmm yes I can try this. It was due to be delivered today but I missed the delivery so I have to pick it up from the post office, however I'm working away next week so I won't be able to test it till I'm back for the weekend

loggio, 2019-08-09:
Just disconnecting, trying over and over to connect. I can see now it will connect, flash a quadrant, and then disconnect... Sometimes reconnecting again a second later only to disconnect again.

Ryzee119, 2019-08-09:
[ogx360_issue10_2.zip](https://github.com/Ryzee119/ogx360/files/3486046/ogx360_issue10_2.zip)
No worries, sorry to be bothering with different files but you could also try this one to see if it makes a difference

loggio, 2019-08-09:
So the symptoms have changed slightly, it's an improvement... Maybe the timing needs to be tweaked beyond default ms drivers?

loggio, 2019-08-09:
No difference

Ryzee119, 2019-08-09:
thanks again, I might see if I can source one of these controllers or ask around if anyone can test their ogx360 with that revision.

loggio, 2019-08-09:
And you're not bothering me mate, happy to be testing. You can keep sending me files

loggio, 2019-08-09:
No way! It just synced.

I waited for a quadrant to flash and then quickly pushed a button and it locked on and stayed connected.

Once I turn the controller off and back on  it struggles to connect again

Ryzee119, 2019-08-09:
Does it help if you're moving an analog stick around whilst it is syncing?
This forces the controller to send interrupt transfers and might help it connect

loggio, 2019-08-09:
I haven't been able to get it to sync again. I can try this again

loggio, 2019-08-09:
Didn't help...

loggio, 2019-08-09:
In case you're interested
This is the dongle I purchased.
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com.au%2Fulk%2Fitm%2F322228480691

loggio, 2019-08-09:
Also,

If you're interested, I've found 3 ebay listing with the controller. 1# finishing later today. All buy it now.

1. https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com.au%2Fulk%2Fitm%2F264316811936

2. https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com.au%2Fulk%2Fitm%2F143347474356

3. https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com.au%2Fulk%2Fitm%2F283557924170

Ryzee119, 2019-08-09:
Awesome, #1 ordered!

loggio, 2019-08-09:
Just make sure you ask the seller to make sure the model number is exactly as pictured!!!

I just noticed he has a max quantity of 2... Meaning he's selling more than one. 

I've had it where the item they send is different to the one pictured.

loggio, 2019-08-10:
Ok, I'm very happy to report that I just picked up my official Microsoft adapter (post office is open for a few hours) and all the controllers sync up perfectly and instantly!

So it is definitely the cheap adapter. 

Would still be good to continue troubleshooting this as I'm guessing in the future a lot of people will have no choice but to buy cheap adapters as the official ones are no longer available.

Ryzee119, 2019-08-10:
Fantastic news, I have this version of controller enroute and a few different receivers to try. If I can replicate the issue I should be able to solve it!

Ryzee119, 2020-12-14:
The XBOXRECV driver has been reworked significantly. If anyone had this issue please retry and report back.

Will close issue in a few weeks if no reports. Reopen if still present :)

closed 2021-01-01 by Ryzee119