Ryzee119/ogx360 #11: 8Bitdo Adapter not working

CroCop86, 2019-08-10:
I have 2 orange and 2 grey 8Bitdo receiver and i can't get anyone to work. Can you help me please somehow?

I have attached the logfile from USB Descriptor Dumper as a textfile

[8Bitdo Receiver.txt](https://github.com/Ryzee119/ogx360/files/3489209/8Bitdo.Receiver.txt)

Ryzee119, 2019-08-11:
Hello,
Interestingly all the descriptor dumps provided look like they're  from a grey adaptor. Not an orange one.

Are these descriptor dumps being  taken from a windows machine?
The Orange adaptors traditionally, enumerate as a Xbox360 wired controller,
The greys enumerate as a generic Playstation controller.

I think I will need to pickup one of these grey units and add support for it.

Ryzee119, 2019-08-11:
You could try udating your orange 8bitdo device to the latest firmware https://support.8bitdo.com/firmware-updater.html

The grey unit is significantly different, so is not currently supported

CroCop86, 2019-08-11:
Thank you for your fast answer. I have updated all four to the latest firmware through the universal 8bitdo updater. I will try to flash them with their own updater:
8BitDo_USB_RR_Firmware_V1.26
8BitDo_USB_RR_FOR_PS_Firmware_V1.26

And maybe downgrade them. If you could add support for the grey ones too, would be really appreciate :)

CroCop86, 2019-08-11:
I have downgrade from Firmware v1.27 to v1.26, now the orange receiver work both. I have attached again the log File for you, so you can see what changed after the downgrade.

Thank you again for your help :)
[8Bitdo Orange after downgrade to FW 1.26.txt](https://github.com/Ryzee119/ogx360/files/3489730/8Bitdo.Orange.after.downgrade.to.FW.1.26.txt)

Ryzee119, 2019-08-11:
That log file looks much better.
So I just did some testing and it is actually possible to write the firmware for the grey unit onto the orange unit (causing incompatibility with the ogx360).
The auto update will then just assume the grey unit and auto update accordingly (and incorrectly).

I would assume now that if you run the auto update tool now that you've written the correct firmware it will work as expected.

I wonder if somehow the firmware got mixed up at some point

CroCop86, 2019-08-11:
I'm not sure how it got mixed up, but for the first time i tried the beta Firmware 1.27 and still the orange icon was shown above the updater. Maybe that was the problem. Downgrading to 1.26 for the orange unit resolved it however. Are you still planing to add support for the grey units?

Ryzee119, 2019-08-11:
I have ordered a grey unit so will see what I can do in the upcoming weeks

Ryzee119, 2019-08-11:
I will mark this issue as closed but will link existing issue #5 for grey 8bitdo adaptor support.

closed 2019-08-11 by Ryzee119