Ryzee119/ogx360 #12: 404 on bill of materials usb controller link

RobertDaleSmith, 2019-08-13:
Thanks for the awesome work here. Looking into building my own unit and noticed one of the Digi-Key links is dead. Would this be a good alternative to the usb host controller?

https://www.digikey.com.au/product-detail/en/maxim-integrated/MAX3421EEHJ-T/MAX3421EEHJ-TTR-ND/1455882

Ryzee119, 2019-08-14:
Hello,

Looks good! However I suggest his link. The same part but allows single units to be purchased and is in stock.

https://www.digikey.com.au/product-detail/en/maxim-integrated/MAX3421EEHJ-T/MAX3421EEHJ-TCT-ND/4895460

I will update the parts list to fix the 404. Thanks for highlighting the issue.

Cheers

Ryzee119, 2019-08-14:
Digikey quick cart link seems to work ok still
https://www.digikey.com.au/short/pbnnjz

RobertDaleSmith, 2019-08-15:
This may seem redundant but.. Does the OGX360 support wired OG Xbox controllers connected with a Xbox controller to USB adapter? I remember years ago making a cable to use my original Xbox controller on a PC, so I assume it's possible.

The reason I ask is because I am brainstorming ideas for installing the OGX360 internally. I thought it would be neat to hack this in between the internal controller ports connector and the xbox's motherboard. Passing the front controller ports into a USB hub and then into the OGX360 and then OGX360 directly to the motherboard connections.

Ryzee119, 2019-08-28:
ogx360 doesnt support OG Xbox controllers currently. There exists a few options out there which have internal installations which use switching relays so that the front ports are still fully functional.
I will close this issue as I have resolved it in commit dd7d544.
Please raised another as an enhancement regarding og controllers.

closed 2019-08-28 by Ryzee119