Ryzee119/ogx360 #13: Third party x360 wireless adapter

CroCop86, 2019-09-11:
Hey Ryzee,

i bought a new third party wireless receiver over ebay and is working on my windows 10 pc. Hower, i can't get it to work with ogx360, it just won't sync. Can you have a look at it please?

[X360 Wireless Receiver.txt](https://github.com/Ryzee119/ogx360/files/3601383/X360.Wireless.Receiver.txt)

Ryzee119, 2019-09-11:
That receiver should work.
Does the LED on the receiver blink at all when holding the sync button?
Where did you get the ogx360 from and is it running the latest firmware?  https://github.com/Ryzee119/ogx360/releases

mspangler99, 2019-11-12:
Do you have any more ogx360 available so I can add Wireless Xbox 360 Controller Support to Your Original Xbox