Ryzee119/ogx360 #15: Another issue sycning with cheap third party adapter

caeberos, 2019-09-19:
I have a kinda crappy xbox 360 wireless adapter that I bought off ebay, it works perfectly when hooked up to my computer, but never quite syncs when I use it on the OGX360, I have three different controllers and none of them sync.    I have included the descriptor dump, and I have also flashed the OGX360 with the 2nd master hex dump included in the issue 10 thread, which didn't help.  If there is a pretty good 3rd party adapter I can get off ebay I wouldn't mind just ordering that to use it, I don't really want to spend $50 on the original Microsoft one, but I will if I need to :)

[DescriptorDump_XBOX360_cheap.txt](https://github.com/Ryzee119/ogx360/files/3628957/DescriptorDump_XBOX360_cheap.txt)

caeberos, 2019-09-19:
so slight correction, the controllers only sync when I am using linux, it remains an unknown device under windows, I am not planning on using this hooked up to my computer anyway, just thought it was worth mentioning

Ryzee119, 2019-09-20:
Thanks ill have a look at this soon. Theres alot of combinations of these clone adaptors and this just looks like another one.

caeberos, 2019-09-21:
thank you so much!  I am glad you are looking into this.  I can't wait to beat halo!!!!

Ryzee119, 2019-09-21:
Hi caeberos, 
Please try version 1.2 here: https://github.com/Ryzee119/ogx360/releases

You can update it as per the instructions here: https://github.com/Ryzee119/ogx360/tree/master/Firmware#programming

Please you note you will ONLY need to update the master module. (I.e the  one closest to the big USB connector)

caeberos, 2019-09-22:
Works Perfectly!   I have only attempted one controller, but it syncs instantly, and also the Steel Batallion mode works perfectly as well!  Thank you so much!

closed 2019-09-22 by Ryzee119