Ryzee119/ogx360 #19: Mad Catz Street Fighter IV SE Fighting Stick

thehijjt, 2019-12-01:
I got excited when I saw an update with a madcatz id, unfortunately it didn't match up with my fight sticks, and the green ring just went dim. I attached the USB info below:

 idVendor=1bad, idProduct=f018, bcdDevice= 4.90
New USB device strings: Mfr=1, Product=2, SerialNumber=3
Product: Street Fighter IV FightStick
Manufacturer: Mad Catz, Inc.
SerialNumber: 0E7D6FB7

input: Mad Catz Street Fighter IV SE Fighting Stick

Ryzee119, 2019-12-02:
Can you provide the rest of the descriptor?
You can use a tool like this:

https://www.thesycon.de/eng/usb_descriptordumper.shtml

Cheers

thehijjt, 2019-12-02:
Information for device Street Fighter IV FightStick (VID=0x1BAD PID=0xF018):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x000F
Current configuration value: 0x01
Number of open pipes: 7

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0xFF	bDeviceClass      (Vendor specific)
0xFF	bDeviceSubClass   
0xFF	bDeviceProtocol   
0x08	bMaxPacketSize0   (8 bytes)
0x1BAD	idVendor
0xF018	idProduct
0x0490	bcdDevice
0x01	iManufacturer   "Mad Catz, Inc."
0x02	iProduct   "Street Fighter IV FightStick"
0x03	iSerialNumber   "0E7D6FB7"
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x0099	wTotalLength   (153 bytes)
0x04	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
0xFA	bMaxPower      (500 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0x5D	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x00	iInterface

Unknown Descriptor:
------------------------------
0x11	bLength
0x21	bDescriptorType
Hex dump: 
0x11 0x21 0x10 0x01 0x03 0x25 0x81 0x14 0x03 0x03 
0x03 0x04 0x13 0x02 0x08 0x03 0x03 

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x04	bInterval         (4 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x08	bInterval         (8 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x04	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0x5D	bInterfaceSubClass   
0x03	bInterfaceProtocol   
0x00	iInterface

Unknown Descriptor:
------------------------------
0x1B	bLength
0x21	bDescriptorType
Hex dump: 
0x1B 0x21 0x10 0x01 0x01 0x01 0x83 0x40 0x01 0x04 
0x20 0x16 0x85 0x00 0x00 0x00 0x00 0x00 0x00 0x16 
0x05 0x00 0x00 0x00 0x00 0x00 0x00 

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x83	bEndpointAddress  (IN endpoint 3)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x02	bInterval         (2 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x04	bEndpointAddress  (OUT endpoint 4)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x04	bInterval         (4 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x85	bEndpointAddress  (IN endpoint 5)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x40	bInterval         (64 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x05	bEndpointAddress  (OUT endpoint 5)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x10	bInterval         (16 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x02	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0x5D	bInterfaceSubClass   
0x02	bInterfaceProtocol   
0x00	iInterface

Unknown Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
Hex dump: 
0x09 0x21 0x10 0x01 0x01 0x22 0x86 0x07 0x00 

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x86	bEndpointAddress  (IN endpoint 6)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0020	wMaxPacketSize    (1 x 32 bytes)
0x10	bInterval         (16 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x03	bInterfaceNumber
0x00	bAlternateSetting
0x00	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0xFD	bInterfaceSubClass   
0x13	bInterfaceProtocol   
0x04	iInterface   "Xbox Security Method 3, Version 1.00, © 2005 Microsoft Corporation. All rights reserved."

Unknown Descriptor:
------------------------------
0x06	bLength
0x41	bDescriptorType
Hex dump: 
0x06 0x41 0x00 0x01 0x01 0x03 

Microsoft OS Descriptor:
------------------------------
0x12	bLength
0x03	bDescriptorType
Hex dump: 
0x12 0x03 0x4D 0x00 0x53 0x00 0x46 0x00 0x54 0x00 
0x31 0x00 0x30 0x00 0x30 0x00 0x90 0x00 

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "Mad Catz, Inc."
0x02   0x0409  "Street Fighter IV FightStick"
0x03   0x0409  "0E7D6FB7"
0x04   0x0409  "Xbox Security Method 3, Version 1.00, © 2005 Microsoft Corporation. All rights reserved."

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
Generic USB Hub
Street Fighter IV FightStick (VID=0x1BAD PID=0xF018) Port: 2

Running on: Windows 10 or greater (Build Version 18362)

Brought to you by TDD v2.12.0, Aug  2 2019, 16:35:05

Ryzee119, 2019-12-02:
Thanks mate. Shouls be able to add it. Will let you know when i get around to it

Ryzee119, 2020-03-15:
Support hopefully added in V1.4 release
https://github.com/Ryzee119/ogx360/releases/tag/1.4

If anyone has this controller and can confirm it works, please report here

Ryzee119, 2020-07-03:
No report for 3 months.
Assuming all is working. Re-open a new issue if an issue is encountered with this controller.

closed 2020-07-03 by Ryzee119