Ryzee119/ogx360 #27: Firmware update not working - Windows 10.

mrDevGit, 2020-06-29:
Hi,

I am trying to flash the firmware that will enable SB support.

However, when I connect the master board via USB it pops up as an "Unknown Device" on a WIndows 10 laptop.

It never shows up as a COM Port.

How do I resolve this?

Thanks

mrDevGit, 2020-06-30:
I thought the reset was just to give you another shot but you have to press reset really quickly after connecting via USB.

Works perfectly.

closed 2020-06-30 by mrDevGit

Ryzee119, 2020-06-30:
Im glad you worked it out. Apologies for not getting back to you!
You can also try double pressing the RESET button in quick succession.
Some of these Arduinos come with slightly different bootloaders and the exact process to get into the bootloader changes slightly.