Ryzee119/ogx360 #28: MadCatz BeatPad and Konami Dance Dance Revolution Pad Support Requested Please

chaosbuzz, 2020-07-02:
Just received the OGX360 and love it so far, was primarily planning on using it to play Ultramix with my Precision Dance Pad Omega 4x (which uses a MadCatz beatpad as it's controller) and Konami official dance pads.  Neither seem to work however.  I am including the full USB descriptors below, hope to see it in the next firmware version and thank you for your amazing work Ryzee!

[DescriptorDump_MadCatz_Beat_Pad.txt](https://github.com/Ryzee119/ogx360/files/4867021/DescriptorDump_MadCatz_Beat_Pad.txt

[DescriptorDump_Konami_Dance_Pad.txt](https://github.com/Ryzee119/ogx360/files/4867023/DescriptorDump_Konami_Dance_Pad.txt)
)

Ryzee119, 2020-07-03:
Thanks for this,
I started making a PR for this:
https://github.com/Ryzee119/ogx360/pull/29

If you know how to git checkout this PR/compile the code you're welcome to try it out. 
I wont be able to test for a few days which id rather do before a new release

Cheers
Ryan

chaosbuzz, 2020-07-03:
I'm in no rush and don't want to muck it up with my lack of knowledge on
the issue, thank you so much for your help!

chaosbuzz, 2020-07-24:
I finally was able to compile the firmware and flash the board which I think I did correctly because the komani xbox 360 dance pad now is recognized and works with OG Xbox!   The Mad Catz/precision dance pad one doesn't though, none of the buttons on the pad seem to work.  The precision dance pad connects through the madcatz via a handmade ethernet.  It works fine on XBOX 360, but doesn't seem to be working here.  I am attaching a photo of what the setup looks like.  The madcatz controller ring appropriately lights up for controller one but doesn't do anything.  You can see the dance pad here: http://www.precisiondancepads.com/en/dance-pads/8-omega-gx.html

Question I have is, if the other komami pad now works, does that mean I compiled the firmware correctly where both should work?  Anything I can try to help make it work?  The precision dance pad is my main DDR platform, so more than anything, hoping to get that to work!  Appreciate your help!
[image]
[image]

Ryzee119, 2020-08-23:
Im not too sure why it wouldnt be working, except that it is not a standard xpad device. (Most of these peripherals behave just like Xbox360 controllers)

You could try these binaries:
https://github.com/Ryzee119/ogx360/actions/runs/220470157

chaosbuzz, 2020-08-25:
Thanks, I'll check it out

Ryzee119, 2020-12-14:
Will close due to inactivity. Please try latest firmware and if these still dont work, please open a new issue :)

closed 2020-12-14 by Ryzee119