Ryzee119/ogx360 #30: Multiple 8bitdo's Freezing

joshuaziel, 2020-07-07:
I'm looking to use multiple 8bitdo's by leveraging a USB hub connected to the single USB port on the OGX360.  I've paired two "gray" 8bitdo's with the "orange" firmware with two Xbox One wireless controllers and am able to use both controllers normally for a period of time.  However, after a few minutes of playing, one of the controllers ceases to provide input and "freezes" on whatever was last inputted.  Sometimes it occurs on Player 1, sometimes it occurs on Player 2.  For example, in a driving game, it could be constantly turning one driver to the left with the gas applied.  No other buttons are effective in resetting this.  The other player is not affected.  To fix, I need to unplug the physical controllers from OGX360 to the Xbox (removing power to all Arduino's) and then reconnect.  Any thoughts?

Ryzee119, 2020-07-08:
Thanks for raising the issue.
Its certainly a less tested use case. 
Is your usb hub externally powered or powered by the ogx360?

If youre able to, could you try with an externally powered usb hub and let me know

joshuaziel, 2020-07-10:
I only have a few bus powered hubs but let me see if I can find a powered one to borrow.  Thank you.

joshuaziel, 2020-07-10:
I should note that I did try it one bus powered hub (AmazonBasics) and it wouldn't even power one 8bitdo.  I moved to an Aukey one and it seems be powering both 8bitodo's fine, save for this issue.  I have had no issues using either one 8bitdo plugging directly into the ogx360 (already a tested use case) nor plugging one 8bitdo into the Aukey bus powered hub and then plugging the hub into the ogx360.  How tight are the margins on the overall circuitry given all of this would have to fit within the Controller 1 power budget?

Ryzee119, 2020-07-14:
I wouldn't think powering all four 8bitdos would be an issue at all really.

This is an interesting problem, and unfortunately may come down to the limits of the atmega32u4s and possibly https://github.com/Ryzee119/ogx360/issues/7.

We may not be meeting timing requirements through the USB hub :(

joshuaziel, 2020-08-12:
I was able to procure an AC powered USB hub and have yet to experience the issue (hard to extensively test two players during quarantine!) so I would think you can close this with guidance to use externally powered hubs instead of bus powered.

Ryzee119, 2020-08-21:
Thank you for that testing :)
I will add a note to the README!

Ryzee119, 2020-08-21:
Added a note

closed 2020-08-21 by Ryzee119