Ryzee119/ogx360 #34: Compatibility with different adapters

vrtlspd, 2020-09-20:
It would be nice if you could also make it compatible with this Ipega PG-9132 adapter. It has a usb port and removes the need of going to a PC for synchronizing for example a PS3 controller.

https://www.aliexpress.com/i/33041731401.html

Ryzee119, 2020-09-27:
Can you please provide a usb decriptor dump of this device

vrtlspd, 2020-09-29:
Is this what you need?

`Information for device PC/PS3/Android Gamepad (VID=0x2563 PID=0x0575):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device address: 0x0005
Current configuration value: 0x01
Number of open pipes: 2

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x2563	idVendor
0x0575	idProduct
0x0100	bcdDevice
0x01	iManufacturer
0x02	iProduct     
0x00	iSerialNumber
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0xFA	bMaxPower      (500 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0111	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0089	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x0A	bInterval         (10 frames)

Microsoft OS Descriptor:
------------------------------
0x2E	bLength
0x03	bDescriptorType
Hex dump: 
0x2E 0x03 0x50 0x00 0x43 0x00 0x2F 0x00 0x50 0x00 
0x53 0x00 0x33 0x00 0x2F 0x00 0x41 0x00 0x6E 0x00 
0x64 0x00 0x72 0x00 0x6F 0x00 0x69 0x00 0x64 0x00 
0x20 0x00 0x47 0x00 0x61 0x00 0x6D 0x00 0x65 0x00 
0x70 0x00 0x61 0x00 0x64 0x00 

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0053 0x0000 0x0000 0x0000 0x0000 0x0000 0x0000 0x0000 0x0000 0x0000 
               0x0000 
0x01   0x0053  "ShanWan    "
0x02   0x0053  "PC/PS3/Android Gamepad"

------------------------------

Connection path for device: 
Controlador Anfitrião Melhorado USB da Família Intel(R) ICH10 - 3A3C
Root Hub
Generic USB Hub
PC/PS3/Android Gamepad (VID=0x2563 PID=0x0575) Port: 1

Running on: Windows 7 with Service Pack 1

Brought to you by TDD v2.15.0, Jun  8 2020, 17:18:07
`

vrtlspd, 2020-10-22:
do you think it can be done?

Ryzee119, 2021-08-10:
ogx360 as this point is only targetting xinput compatible controllers. Whilst this would be possible, it is difficult to handle all these generic HID controllers. Particularly with the limitations of the Atmega32u4 chip.

It is currently outside the scope of those project.

closed 2021-08-10 by Ryzee119