Ryzee119/ogx360 #35: Hyperkin Duke compatibility

K-Hobert, 2020-09-27:
Whenever my Hyperkin Duke is plugged in , directly to USB port on the Ogx360 the Hyperkin vibrates constantly, no input detected and OLED remains off. Same when used over a powered USB hub. Latest firmware 1.4 also tried compiled from master with  same result.

Ryzee119, 2020-09-27:
Thanks for reporting the issue. I'll check it out when I can

Ryzee119, 2020-09-27:
Just tested in my end. It vibrates for 1-2 seconds then works ok. 🤔 

There's this delay here that could be tweaked:
https://github.com/Ryzee119/ogx360/blob/74a12a085fc44647db9afb5726fbe0ab6892c1a1/Firmware/ogx360_32u4/lib/USB%20Host%20Shield/XBOXONE.cpp#L169

Ryzee119, 2020-09-27:
I did just publish a v1.5, but it's probably the same as your manually compiled version so may not help (could try it anyway)

K-Hobert, 2020-09-27:
Thank you. I’ll test 1.5, wonder if they revised  the controller at any point? Looked at PID and VID of mine no change there. Could it be a power issue, is a powered hub required for the Hyperkin?

Ryzee119, 2020-09-27:
I wasn't using a powered hub, but it could be.

K-Hobert, 2020-09-28:
I don’t believe power is the issue, will test further later in the week...I’m thinking Hyperkin may have revised the controller model is M01668/1721P08. Tried playing around with delay functions, still noting :/

Ryzee119, 2020-09-28:
thanks for trying that. oof hopefully they havent added something too weird. Can you provide a full descriptor dump of the controller?
I have used https://www.thesycon.de/eng/usb_descriptordumper.shtml on windows before. Might be something obvious when compared to mine.

Finally, if you want to get adventurous you can use Wireshark to record USB traffic during enumeration for me to review.

K-Hobert, 2020-09-28:
Bit of amateur hour for me here...Hyperkin Duke version 1.01 changed the PID. It’s 0x1618 compiled and flashed it works.

Ryzee119, 2020-09-28:
Awesome!
Just to confirm v1.01 has PID `0x1618` instead of `0x0652`?

K-Hobert, 2020-09-28:
Yes:)

Ryzee119, 2020-09-28:
Thank you for the issue report. Ive added this combination into master branch 👍 https://github.com/Ryzee119/ogx360/commit/27942f1965dc4b0a7e773592d668ba88c2513541

closed 2020-09-28 by Ryzee119