Ryzee119/ogx360 #38: Mayflash adapters.

K-Hobert, 2020-10-01:
The Mayflash magic pro is a USB crossover adapter which supports x-input. Displays its self as a “Xbox 360 controller for windows”. The interest for myself is input usb on the adapter itself. Could be used to support more wired controllers that lack xinput and Bluetooth. Having tested the Bluetooth connectivity it worked seemly perfect. But using the device bugged out the board other wired controllers (on an externally) powered hub stopped receiving input to the console. Board has to be powered off but the Mayflash keep working....added the VID/PID into 360 support column.

K-Hobert, 2020-10-01:
Figured a work around, first plug the mayflash in first (synced to a wireless controller) then use a second controller as a dummy controller. Plug a 3rd controller in the 3rd acts as the playable second controller.

Ryzee119, 2020-10-01:
interesting work around! 

Can you provide a USB descriptor dump of this deviec when connected into windows? Im wondering if it's a composite device and registering a 2nd slot but not sending outputs to it.

K-Hobert, 2020-10-06:
Information for device MAGIC-S PRO (VID=0x2F24 PID=0x00F7):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0001
Current configuration value: 0x01
Number of open pipes: 2

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0xFF	bDeviceClass      (Vendor specific)
0xFF	bDeviceSubClass   
0xFF	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x2F24	idVendor
0x00F7	idProduct
0x0101	bcdDevice
0x01	iManufacturer   "."
0x02	iProduct        "MAGIC-S PRO"
0x03	iSerialNumber   "0243ED2700009D45"
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x0030	wTotalLength   (48 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0xFA	bMaxPower      (500 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0xFF	bInterfaceClass      (Vendor specific)
0x5D	bInterfaceSubClass   
0x01	bInterfaceProtocol   
0x00	iInterface

Unknown Descriptor:
------------------------------
0x10	bLength
0x21	bDescriptorType
Hex dump: 
0x10 0x21 0x10 0x01 0x01 0x24 0x81 0x14 0x03 0x00 
0x03 0x13 0x02 0x00 0x03 0x00 

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x04	bInterval         (4 frames)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x08	bInterval         (8 frames)

Microsoft OS Descriptor:
------------------------------
0x12	bLength
0x03	bDescriptorType
Hex dump: 
0x12 0x03 0x4D 0x00 0x53 0x00 0x46 0x00 0x54 0x00 
0x31 0x00 0x30 0x00 0x30 0x00 0x90 0x00 

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "."
0x02   0x0409  "MAGIC-S PRO"
0x03   0x0409  "0243ED2700009D45"

------------------------------

Connection path for device: 
Intel(R) ICH10 Family USB Universal Host Controller - 3A36
Root Hub
MAGIC-S PRO (VID=0x2F24 PID=0x00F7) Port: 1

Running on: Windows 10 or greater (Build Version 18363)

Brought to you by TDD v2.15.0, Jun  8 2020, 17:18:07

K-Hobert, 2020-10-06:
With a device plugged in to Mayflash's USB port two changes happen in the descriptor
Current configuration value: 0x00
Number of open pipes: 0

Ryzee119, 2021-07-06:
Could you please retry this with latest v2.x firmware

Ryzee119, 2021-08-10:
Will close as stale. If issue is encounted on latest firmware please create a new issue

closed 2021-08-10 by Ryzee119