Ryzee119/ogx360 #40: 8bitdo - Xbox One Controller issue

CrispyWhispy, 2020-10-16:
Hi there, 

Just trying to get the ogx360 working with an Xbox One controller via 8bitdo.

The controller pairs with the 8bitdo, but doesn't seem to be passing commands. 

8bitdo confirmed working with Nintendo Switch. 
Controller works fine natively via Bluetooth to PC.

I have tried 8bitdo firmware versions-
1.26
1.27
1.28
1.30
1.33

The Xbox One controller firmware was well out of date when I started - ver. 3.x.xxxx.x
Now I've updated it to ver. 4.8.1923.0

Just followed your steps in the firmware section - https://github.com/Ryzee119/ogx360/tree/master/Firmware
and updated the Master board to your latest (v15) hex - ogx360_32u4_master.hex

Also, I assuming if I wanted to use the Xbox One controller as wired, I would just connect it via USB to the USB port the 8bitdo/Xbox360 wireless receiver would plug into? This still doesn't work though.

Unfortunately I do not have an Xbox360 wireless receiver to test my 360 controllers with the ogx360.

Any help would be great. Thanks.

Ryzee119, 2020-12-08:
Hi apologies for the late response.
Do you get a second led appears on the master arduino when you plug in the 8bitdo adaptor?
Is it the orange unit and not the grey?
I am working on a separate branch on github here that may improve some of the packet reliability that could help. Should be done in a day or so

CrispyWhispy, 2020-12-09:
Hey, not a drama. I saw you had replied to other threads so I thought to just wait my turn haha.
Here's a photo, using the orange 8bitdo and red LED on the daughter board flashes on and off. 
[image]

Ryzee119, 2020-12-09:
Have you got this working on any device at all?

Flash red led at a rate of 500ms off, 500ms on is a sign that it cannot initialise the MAX3421 🤔

CrispyWhispy, 2020-12-09:
Unfortunately I don't have an Xbox 360 wireless dongle if that's what you mean.
And the using a USB cable to put the controller in "wired mode" through the ogx won't work either.
The single Xbox One (bluetooth) controller that I know works connecting straight to PC, or via 8bitdo to Switch.

In this video below you can see the power up of the device, the 8bitdo gets power, syncs with the controller - no input accepted. The flashing is around 500ms I would say. What's the next step if the MAX3421 is the issue?
I press the 'RESET' button on the motherboard and the two daughterboard LEDs come on, with the originally flashing LED going solid for a few seconds, then flashing again - still no input. 

https://drive.google.com/file/d/1s6TCYU2YNybFQvBiI2Hso1EIs-yx2N3j/view?usp=sharing

Ryzee119, 2020-12-09:
Unfortunately its normally a sign of something faulty. How proficient are you in fault finding? 
Are you located in Aus?

CrispyWhispy, 2020-12-09:
That's a shame. Bought this unit from TEK Nemesis in Canada. 
Pretty proficient from a general standpoint, I'm a Systems Engineer for an ISP by day (and night)....in this regard I suppose I could crack out the multimeter and probe away if you point me in the right direction.
Yeah I'm in QLD.

Ryzee119, 2020-12-09:
You could try this firmware.
Its just alot newer and had lots of changes. It may not may not help.
[MASTER_360WLESS_360W_STEELBATTALION.zip](https://github.com/Ryzee119/ogx360/files/5664868/MASTER_360WLESS_360W_STEELBATTALION.zip)

If thats no good, I have a bunch of spares here. I can shoot one youre way if youre happy to send me the dud?
Ive seen a few of these ones that fail with no apparent reason so I would be interested to try debug it.

CrispyWhispy, 2020-12-09:
Alrightly, happy to send it to you based on the outcome of this -

Just flashed that firmware 
[image]

Now the daughterboard LED flashing every 10 seconds, still not accepting input

Ryzee119, 2020-12-09:
Sounds faulty. Hit me up at wendland@live.com.au to exchange info ill send one you're way.

Keep your case and cables etc.

CrispyWhispy, 2020-12-09:
Not a drama, just emailed you now. Many thanks.

Ryzee119, 2020-12-11:
Was this damn crystal
[image]

closed 2021-01-01 by Ryzee119