Ryzee119/ogx360 #41: Customized ogx360 hardware with Pro Micro 3.3V

phantom-8, 2020-11-18:
I've built a simplified version of ogx360 with a Pro Micro 3.3V/8MHz board and a Mini USB Host Shield 2.0.  It only supports 1 Xbox 360 controller.  Everything seems to work fine, except it missed a button release event once a while.  I suspect the culprit is the slower 8MHz of the Pro Micro.  Do you have any idea on how to tackle the problem?

[image]

Ryzee119, 2020-11-18:
Very nice!
It could certainly be an issue. As its limited to one controller you can try these:

Make this loop to 1 instead of 4 to avoid some overhead
https://github.com/Ryzee119/ogx360/blob/27942f1965dc4b0a7e773592d668ba88c2513541/Firmware/ogx360_32u4/main.cpp#L223

Have you adjusted `F_CPU` to match your new clock? Not sure how your compiling, but its a define setup in your project.

You could try bumping up the SPI speed. I think its set here (and look for other instances for 12000000 in that file. The default was originally 26000000.

https://github.com/Ryzee119/ogx360/blob/27942f1965dc4b0a7e773592d668ba88c2513541/Firmware/ogx360_32u4/lib/USB%20Host%20Shield/usbhost.h#L62

phantom-8, 2020-11-18:
Thank you so much for the prompt reply.  I've already removed unnecessary code in main.cpp.  At the moment, all I2C code are gone; the main "for loop" iterates only once; master's slave handling code are gone.  The F_CPU is already set to 8000000UL.  Unfortunately, the problem still exist.  

I'm gonna try to play around with the SPI setClockSpeed and see if it helps.  Thanks again for your kind assistance.

phantom-8, 2020-11-18:
Just tested the SPI Clock.  In the ushhost.h, the ATMega32u4 doesn't use setClockSpeed.  Instead, the clock speed is being set via SPCR & SPSR registers.
```
78                 SPCR = 0x50; //SPI Enable and MASTER,
79                 SPSR = 0x00; // 0x01 = SPI2X Ryzee119, changed from 0x01 to 0x00 to disable SPI2X. Speed is fclk/4 now.
```
The SPI Clock is currently being set to 4MHz for Pro Micro 16MHz version, and 2MHz for Pro Micro 8MHz version.  I've bumped the clock speed to 4MH by setting SPSR=0x01.  The 4MHz SPI Clock is the maximum that can be set on the Pro Micro 8MHz version.  Unfortunately, this didn't fix the problem.

Can I ask how the MAX3421 handles controller button events?  Does it only send button change events via SPI (that is, only send 1 event for button down and another event for button up)?

Ryzee119, 2020-11-18:
ah yes ofcourse. been a while since Ive looked deep into this code.

MAX3421 generates an interrupt on all new events on the usb device. ogx360 polls this interrupt register to detect changes and applies them accordingly.

Most controllers will only send a new packet if a change has been detected. For example if you're moving the analog stick around do the missed button presses happen?

Im wondering if this section is happening *too quickly* and is clearing previous buttons before it can send it to the og xbox. ogxbox is capped as 4ms per interrupt packet. xbox 360 can do 1ms.

https://github.com/Ryzee119/ogx360/blob/27942f1965dc4b0a7e773592d668ba88c2513541/Firmware/ogx360_32u4/main.cpp#L230-L238

Although counter intuitive, add a 4ms delay on each loop.

The 4ms thing is here. You could try changing the `>=4` to 0 or 1. Although I have had issues when im spamming the og too fast much. :)
https://github.com/Ryzee119/ogx360/blob/27942f1965dc4b0a7e773592d668ba88c2513541/Firmware/ogx360_32u4/main.cpp#L688

phantom-8, 2020-11-19:
> MAX3421 generates an interrupt on all new events on the usb device. ogx360 polls this interrupt register to detect changes and applies them accordingly.
> 
Thanks for the explanation.  

> Most controllers will only send a new packet if a change has been detected. For example if you're moving the analog stick around do the missed button presses happen?
> 
Yes, if I moved the analog stick, the missed button press/release will be updated.

> Im wondering if this section is happening _too quickly_ and is clearing previous buttons before it can send it to the og xbox. ogxbox is capped as 4ms per interrupt packet. xbox 360 can do 1ms. [...]
It seems to me your code is fine.  I don't think the problem is related to your button handling or the timing on sending HID report to OG.  My theory is that the 8MHz Pro Micro is only running at half speed, it's not fast enough to poll the interrupt register to capture all button change events.  I think the following are the possible ways to solve the problem:

1) Use a Pro Micro version with 3.3V/16MHz which are being sold on ebay & Aliexpress.  The ATMega32u4 will be running out of spec, but most people reports are positive.

2) The current implementation uses polling to process the button events.  This will be problematic for slower MPU.  A better way would be interrupt driven.  I think you mentioned to migrate to UHS3.0 in the future.

3) UsbHost.Task() is not being called fast enough and result in lost events.  Currently, it's being called once at the beginning of the main loop.  We can try to add 1 or more calls to UsbHost.Task() in the main loop and spread them evenly. 

I will try approach 3 first and see if it helps.  If not, I think I may go with approach 1.

phantom-8, 2020-11-19:
Approach 3 didn't work as the problem is within UsbHost.Task().  The Poll() function makes calls to checkControllerPresence(), checkControllerBattery(), setLedRaw(), chatPadKeepAlive1() & chatPadKeepAlive2() every 2.5 seconds.  These functions will send commands to MAX3421 via SPI bus.  As long as I comment out all these functions, the system will run reliably and no more message lost.  If I leave any 1 of these functions intact, it will also result in message lost.  These functions create more traffic on the SPI bus and the 8MHz Pro Micro seems insufficient to handle it.  I guess my only option now is to use a Pro Micro version with 3.3V/16MHz.

https://github.com/Ryzee119/ogx360/blob/27942f1965dc4b0a7e773592d668ba88c2513541/Firmware/ogx360_32u4//lib/USB%20Host%20Shield/XBOXRECV.cpp#L280-L315

jimnarey, 2020-11-24:
Hi Phantom8 - is there any chance you could share your simplified code? I was about to dive in and do something similar myself (so far using a Leonardo with a full sized shield).

Thanks.

Ryzee119, 2020-11-24:
https://github.com/Ryzee119/ogx360_t4
I knocked this up recently.
Which uses a single teensy4.1 that can do dual usb. 
Its a huge jump in power and flash memory so opens up lots of possibilities for future work.

Ryzee119, 2020-11-24:
[image]

jimnarey, 2020-11-24:
And based in VS Code! Thank you so much for pointing me to this.

phantom-8, 2020-11-25:
I've built another version based on Pro Micro 5V/16MHz.  I used 5V to 3.3V logic convertors for SS, MOSI, SCLK & D9.  Basically, the circuit is the same as the one on your schematics.  Unfortunately, this version also encounters button lost events, exactly like the 3.3V/8MHz version.  I tried using both your Github compiled firmware and also my own built.  They've the same problem.  I'm really puzzled on what's going on.  

@Ryzee119, is it possible for you to test whether your version has button lost events?  The way that I tested was using the Controller test tool from your Video.  I just keep pressing/releasing the same button and watch the controller screen for missing press/release.  In average, I would lose a button event every 100 button presses.

@jimnarey, do you still need the trimmed down version from me?  Let me know if you do and I'll cleanup the source code and upload to Github as a fork.

Ryzee119, 2020-11-25:
Quickly have a shot if this
[ogx360_32u4_master.zip](https://github.com/Ryzee119/ogx360/files/5594796/ogx360_32u4_master.zip)

Ryzee119, 2020-11-25:
Does it happen on D-Pad and ABXY actually?

Ryzee119, 2020-11-25:
Desregard previous, try this one
[ogx360_32u4_master.zip](https://github.com/Ryzee119/ogx360/files/5594835/ogx360_32u4_master.zip)

phantom-8, 2020-11-25:
> Desregard previous, try this one
> [ogx360_32u4_master.zip](https://github.com/Ryzee119/ogx360/files/5594835/ogx360_32u4_master.zip)

Do u want me to try this version?

I forgot to mention that I was using a Xbox 360 Wireless controller with Chatpad.  I believe it will be easier to reproduce with Chatpad connected.

Ryzee119, 2020-11-25:
ooof yea you forgot to mention that :P Is this for steel battalion?

phantom-8, 2020-11-25:
> Do u want me to try this version?
>
> I forgot to mention that I was using a Xbox 360 Wireless controller with Chatpad. I believe it will be easier to reproduce with Chatpad connected.

> ooof yea you forgot to mention that :P Is this for steel battalion?

Yeah, it's for Steel Battalion.  BY THE WAY, it also happened with Chatpad when I was using Pro Micro 3.3V/8MHz.  I've test without Chatpad on 5V/16MHz version yet.

Ryzee119, 2020-11-25:
All good I can replicate now. Ill see if I can fix

phantom-8, 2020-11-25:
> All good I can replicate now. Ill see if I can fix

Great!  I thought I was doing something wrong.  :)

Ryzee119, 2020-11-26:
Im struggling to fix this. Its almost like the chatpad packets are overwriting the button packets. i dont know know. This could be an inherent issue with Usb Host Shield v2 :(

phantom-8, 2020-11-26:
When I was testing the old Pro Micro 3.3V/8MHz, I also experienced button event lost even without chatpad connected.  However, the problem occurrence is much less frequent without chatpad.  I will do some more testing on the 5V/16MHz version to confirm whether the problem also existed without chatpad.

phantom-8, 2020-11-26:
I've just tested the 5V/16MHz version with a Xbox 360 Wireless Controller without chatpad connected.  The button event lost also occurred, but it's very difficult to reproduce and took me 7 minutes to get 1 lost event.  To make it easier to reproduce, you can modify the timer in the Poll() function from 2500ms to let's say 200ms.  Then, the problem will occur much more frequent.  From my experience, as long as you send any command to the controller over SPI bus in the Poll() function, there is a possibility that a button event may be lost.

https://github.com/Ryzee119/ogx360/blob/27942f1965dc4b0a7e773592d668ba88c2513541/Firmware/ogx360_32u4/lib/USB%20Host%20Shield/XBOXRECV.cpp#L280-L286

phantom-8, 2020-11-27:
I'm thinking whether it's possible to manually request a HID report from the controller.  If it's possible, maybe a timer can be added to regularly request additional HID report in order to bypass the missing button event issue.

Ryzee119, 2020-11-27:
Yea it mighy be possible to request button state. Not sure of the command to do that yet

phantom-8, 2020-11-27:
I searched the net, but couldn't find the manual request button state command.  If you happen to find it, please post here.  I can try to add the code and test if it works.

jimnarey, 2020-11-28:
@phantom-8 - I've pared back the code myself, but thank you! You can see the repo here if you're interested: https://github.com/jimnarey/SimpleXboxControllerAdapter

phantom-8, 2020-11-29:
@jimnarey, excellent work!  You've provided a very detailed description on how to build a simple version with off-the-shelf parts.  My interest is to use a Xbox 360 Wireless controller with Chatpad to emulate Steel Battalion controller.  I've the original full size controller, but will like to play SB LOC with my daughter.  That's why I'll like to fix the message lost problem.  I'm gonna try to use a STM32F4 to test whether there will be any message lost for a faster version of the the MCU.  I suspect the problem is caused by the slow Pro Micro MCU.

jimnarey, 2020-11-29:
Thank you! I'm now making a bit of progress implementing PS3 support but am having some problems compiling the project with PS4 support. There's something a bit strange going on. How are you compiling yours?

phantom-8, 2020-11-29:
I'm compiling under MacOS using CMake & avr-gcc.  I followed instructions here.
https://github.com/Ryzee119/ogx360/pull/32

jimnarey, 2020-11-29:
Thank you. I've had a look at the issue/instructions but I can't see from them how to compile locally. I suspect, though, my problem is something a bit more fundamental as I was able to reproduce it in Microchip Studio. I've opened another issue. It's quite odd, I can add in the PS3 controller libraries perfectly well without the linker complaining.

If you could let me know what you're running on MacOS and you don't mind, I'll add it to my guide. At the moment it's relying on VSCode to run the CMake tasks.

phantom-8, 2020-11-29:
The is a shell script auto-build file under
ogx360/.github/workflows/ci_build.sh

Just copy the script to your main ogx360 directory and execute the script.  It should automically build the firmware .

jimnarey, 2020-12-02:
How are you getting on with the lost button presses? 

With a Leonardo and USB Host Shield (so 5V, 16Mhz) using wired controllers only (Xbox One, 360, PS3, PS4), I haven't noticed any missing button presses yet but testing has basically involved a bit of Halo to check button mapping/for latency and a bit of Burnout 3 to check R1/R2 pressure is being handled properly. I'm happy to do further testing if that would be helpful.

I'm about to build one using a Pro Micro at 5V/16Mhz, 3.3V regulator, level shifter and mini shield. I'll see how that works.

I've been looking back through this discussion and noted the bit about the Poll() function in XBOXRECV. The equivalent in XBOXUSB is much simpler:

uint8_t XBOXUSB::Poll() {
        if(!bPollEnable)
                return 0;
        uint16_t BUFFER_SIZE = EP_MAXPKTSIZE;
        pUsb->inTransfer(bAddress, epInfo[ XBOX_INPUT_PIPE ].epAddr, &BUFFER_SIZE, readBuf); // input on endpoint 1
        readReport();
#ifdef PRINTREPORT
        printReport(); // Uncomment "#define PRINTREPORT" to print the report send by the Xbox 360 Controller
#endif
        return 0;
}

Could you meet your overall aim with a chatpad and wired controller?

phantom-8, 2020-12-02:
I've done some testing with ESP32 which is a lot faster than Pro Micro.  Unfortunately, the lost event also occurred in this version.  Here is what I've found so far:

- My previous statement of sending any commands over SPI bus will cause button event lost is not totally correct.  It's not the SPI send commands that cause button lost.  It's the reply resulted from those commands that cause button event message lost/overwritten.
- In the Poll() function, the following commands will result in replies from Controller.  Only replies with 2 or more consecutive messages will overwrite button events.  Command with a single reply message works fine.
  * checkControllerPresence(): Reply with 2 messages (1 short & 1 long).
  * checkControllerBattery(): Reply with 1 long message
  * setLedRaw(): No reply message
  * chatPadKeepAlive1(): Reply with 2 or more long messages
  * chatPadKeepAlive2(): Reply with 2 or more long messages
- The commands checkControllerPresence(), chatPadKeepAlive1() & chatPadKeepAlive2() will result in 2 or more consecutive reply messages and will cause button event lost/overwritten.  The possibilities of event lost depends on number of messages on the reply.  The more the reply messages, the higher the change of losing the button events.  Therefore, checkControllerPresence() has less chance of overwritting button events comparing to chatPadKeepAliveN().
- In order to reproduce the problem easier, I've change the checkStatusTimer from 2500ms to 10ms; as this will create a lot more reply messages.
- On ESP32, I've changed the SPI speed from 4MHz to 26MHz which is the maximum supported by MAX3421E.  Problem still exist.
- I've also tried to connect 2 wireless controllers.  On controller 1, Pro Micro keeps calling chatPadKeepAliveN() in every 10ms.  On controller 2, Pro Micro won't send any SPI commands and I continuously hit the same button on Controller 2 to test for event lost.  Under this configuration, there are no message lost/overwritten.  Thus, the problem has nothing to do with the bandwidth of the SPI bus, since the same bus still has a lot of reply messages from controller 1.  The difference is that they're on 2 different endpoints.

@jimnarey, the reason the Poll() function in XBOXUSB is much simpler is because it doesn't support Chatpad.  In fact, I'm in the process of adding Chatpad support to the XBOXUSB (Wired Controller).  From my test with 2 controllers above, it's very likely the button event lost/overwritten problem won't occur on XBOXUSB because the Chatpad is using a different endpoint.  As for the Wireless Controller, the same endpoint is being shared by controller button events & chatpad events and this is when messages got lost.

if you wanna simulate the problem, you can simply change the checkStatusTimer from 2500ms to 10ms in the Poll() function and continuously hit the same button.  Then, you should be able to reproduce it.

phantom-8, 2020-12-02:
> I'm about to build one using a Pro Micro at 5V/16Mhz, 3.3V regulator, level shifter and mini shield. I'll see how that works.

I'm still waiting for my Pro Micro 3.3V/16MHz to arrive.  If it works, we don't need level shifters and the build is much easier.  BY THE WAY, I think 3.3V/8MHz works fine as the message lost problem is not caused by the slower MCU.  However, it's nicer to have a faster MCU. :P

Ryzee119, 2020-12-02:
@phantom-8 i think is onto it. 
Ive been trying and I simply cant fix it. I dont know if its a limit of UHS2 or the MAX3421, or we're missing something in our understanding of the protocol all together.
(Or it actually happens on the real xbox 360)

phantom-8, 2020-12-02:
@Ryzee119, I noticed that on ogx360_t4, you're emulating Steel Battalion controller with mouse & keyboard.  Do you have any plans to add support to Xbox 360 Wireless Controller with Chatpad.  I'm interested to know if the message lost problem still exist on the new platform.

Ryzee119, 2020-12-02:
Yes i plan to. Im also interested 😊

phantom-8, 2020-12-03:
@Ryzee119, I noticed there are 2 versions of Xbox 360 Wired Controller, 1.10 & 1.14.  The one I have is 1.14.  The endpoint numbering seems to be different.  In your XBOXUSB.cpp, you're using endpoint address 0x02 for the controller output pipe.  Is your controller version 1.10?  I need to change the output pipe endpoint address to 0x01 for mine to work.

https://github.com/Ryzee119/ogx360/blob/27942f1965dc4b0a7e773592d668ba88c2513541/Firmware/ogx360_32u4/lib/USB%20Host%20Shield/XBOXUSB.cpp#L146-L157

I've gotten the chatpad to respond and it's a pain in the ass to get the timing right.  Will test for missing button events later.

Ryzee119, 2020-12-03:
Yea must be different versions. Mine is definitely pipe 0x02.
Is yours both on pipe 0x01?

The correct way is to parse the configuration descriptor to pull out the endpoint addresses.
It gets the device descriptor here. There another function called `getConfDescr` which I think might work.

Unfortunately these shortcuts exist to save space on the tiny microcontroller. I didnt know there was another version until now

phantom-8, 2020-12-03:
Yeah, both my Controller Input & Output pipes are 0x01.  The Chatpat endpoints are also different for the 2 versions.  Need to add special handling.

Ryzee119, 2020-12-03:
Ive tried get UHS v3 running on the ogx360 which doesnt have any of these shortcuts. Ive got it going but constantly runs out of memory. I have a feeling the limited 32u4 just cant cut it :(

phantom-8, 2020-12-03:
> Ive tried get UHS v3 running on the ogx360 which doesnt have any of these shortcuts. Ive got it going but constantly runs out of memory. I have a feeling the limited 32u4 just cant cut it :(

Is it possible to trim down the code just enough to test the Xbox 360 Wireless controller with Chatpad to see if there are any lost button events on UHS v3?

I've just finished testing for button event lost with XBOXUSB.  There're aboslutely no button event lost, no matter how hard I jam the bus with Chatpad traffic.  This is expected as the chatpad & the controller are on different endpoints.  I still have some work to do to finish up the chatpad & Steel Battalion coding on the XBOXUSB.  When I'm done, I'll create a pull request.

Ryzee119, 2020-12-03:
I have found something interesting!

Try changing this line
https://github.com/Ryzee119/ogx360/blob/27942f1965dc4b0a7e773592d668ba88c2513541/Firmware/ogx360_32u4/lib/USB%20Host%20Shield/XBOXRECV.cpp#L377

to this

`if(readBuf[0] == 0x00 && (readBuf[1] == 0x01 || readBuf[1] == 0x03))`

When mashing the chatpad and buttons at the same time, theres is new packet with a 0x03 ID. I cant see this documented anywhere online.

Seems to have fixed my problems!

Ryzee119, 2020-12-03:
I got the chatpad going of my ogx360_t4 variant and it was doing the same thing. Saw these unusual packets and looked into it.
In hindsight this rings a bell.. damn.

I havent tested this on traditional ogx360 yet. so please let me know

phantom-8, 2020-12-03:
Nice catch!  Microsoft combined the button & chatpad packets into a single one!  In the 1st test, I continuously performed controller button bashing with chatPadKeepAliveN() running every 10ms.  It created lots of 0x03 packets and this is in fact the cause of missing button presses with chatpad ’cos we didn’t check for the 0x03.  lol

Unfortunately, there is a catch.  As I mentioned earlier, besides chatPadKeepAliveN(), checkControllerPresence() will also result in 2 consecutive reply messages.  My 2nd test is to perform controller button bashing with checkControllerPresence() running every 10ms.  This time a button lost event really occurred.  Here is the log:

```
[00:01:17.049] Controller 0: 08 80 
[00:01:17.051] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
[00:01:17.060] Controller 0: 08 80 
[00:01:17.062] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
[00:01:17.066] Controller 0: 00 01 00 F0 00 13 00 10 00 00 E8 00 AD F6 66 10 7C FA 00 00 00 00 00 00 00 00 00 00 00
[00:01:17.071] Controller 0: 08 80 
[00:01:17.073] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
[00:01:17.078] Controller 0: 00 00 00 F0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:17.083] Controller 0: 08 80 
[00:01:17.085] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
[00:01:17.094] Controller 0: 08 80 
[00:01:17.096] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
```

```
[00:01:17.315] Controller 0: 08 80
[00:01:17.317] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
[00:01:17.326] Controller 0: 08 80
[00:01:17.328] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
[00:01:17.337] Controller 0: 00 00 00 F0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[00:01:17.341] Controller 0: 08 80
[00:01:17.348] Controller 0: 08 80
[00:01:17.350] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
[00:01:17.359] Controller 0: 08 80
[00:01:17.361] Controller 0: 00 0F 00 F0 00 CC FD C3 97 D0 74 4C 78 D0 00 20 1B A3 20 1D 30 03 40 01 50 01 FF FF FF
```
At time 00:01:17.066, button A has been pressed and detected.  At 00:01:17:337, you’ll see the message “00 00 00 0F 00…”.  There should be a button release message before this one and it’s missing.  However, the occurrence of this is extremely rare and it’s very difficult to reproduce when calling checkControllerPresence() every 2.5 seconds.  I recalled that it took me 7 minutes to got 1 lost event last time.  Thus, I wouldn’t worry about fixing this one.

Anyway, I’m happy with the Chatpad fix and thank you so much for the hard work!  :)

If you're interested, you can download the above log file for the 2nd test here
https://drive.google.com/file/d/13SH5V3YFz0XbzkvY0QnGK6sydX_0xA3i/view?usp=sharing.

Ryzee119, 2020-12-03:
I made alot of improvements/changed a few things during this investigation in ogx360 code. It may help these last few lost packets. Ill clean it up and push it as soon a I can

phantom-8, 2020-12-04:
Great!  Once you push the changes, I can help to retest.  Thanks again!

Ryzee119, 2020-12-04:
https://github.com/Ryzee119/ogx360/tree/xboxrecv/Firmware/ogx360_32u4
Try this branch. Needs to be burn tested will be interesting to see if this helps with the rare button press you are missing now

phantom-8, 2020-12-05:
In source file XBOXRECV.cpp, you've added the handling of 0x03 packet for button events, but you haven't handled 0x03 on the chatpad.  The 0x03 is a merged packet for controller & chatpad events.  I would suggest change line #373 to this in order to avoid missing chatpad events.

```
    } 
    if(readBuf[0] == 0x00 && (readBuf[1] == 0x02 || readBuf[1] == 0x03)){
```

I'll preform testing in a moment.

Ryzee119, 2020-12-05:
the `0x03` packet I think only contains the chatpad led status, not chatpad button events from what I have seen, but I could be wrong though

phantom-8, 2020-12-05:
It's most likely the Controller firmware merged button & chatpad events due to traffic concern.  If 0x03 can contain chatpad LED events, it's very likely it may also contain chatpad buttons.

Ryzee119, 2020-12-05:
yep looks like youre right :) Ill push a new change shortly

Ryzee119, 2020-12-05:
Alright, ive push those changes. Disclaimer havent tested them myself

phantom-8, 2020-12-05:
Will test the latest update now.

Ryzee119, 2020-12-05:
sorry, just found another small bug which I have fixed related to chatpad returning incorrect buttons if you mashed multiple buttons quick enough  👍 
Pushed that too.

Nice this seems to be working alot better now

phantom-8, 2020-12-05:
No problem.  Will fetch the update.  :)

phantom-8, 2020-12-05:
Time for test report, here are the findings:

1) When pressing the Guide button on the controller to connect to the wireless receiver, they sync up perfectly every time.  However, the controller number LED indicator sometimes keeps flashing (4 ring LEDs keep flashing) without settling to show the controller number.  To reproduce, press Guide to connect.  Then, press and hold Guide for a few seconds to disconnect.  I’ve repeatedly tested 10 times, 4 out of 10 will just keep flashing.  Other 6 times successfully settled to show controller number LED.  BY THE WAY, I’ve re-tested the original firmware before these code changes, this problem wasn’t there.

2) With the stock firmware, that is checkStatusTimer at 2500ms, I’m unable to reproduce button lost.  PERFECT!!!

3) Modified firmware with checkStatusTimer at 10ms and only calls to checkPadKeepAliveN(), no lost button event detected!

4) Modified firmware with checkStatusTimer at 10ms and only calls to checkControllerPresence(), lost button event occurred with button bashing.  With the attached log file, at time 00:00:15:183, controller button A press was deteced.  But no button release event was deteced afterwards.

The checkControllerPresece() still results in lost message under extreme condition with 10ms calling interval.  However, I’m unable to reproduce it under normal condition with statusStatusTimer at 2500ms.  I believe it will be very difficult to encounter under normal usage.  

Test report for test#4 can be download here:
https://drive.google.com/file/d/1CJuaxPktqSTsD7N0J8CRyJ8JSVkMuBa_/view?usp=sharing

Ryzee119, 2020-12-05:
Once again, thank for you for the thorough testing.
I have push some more changes to hopefully fix that LED issue. These wireless controllers are very picky 👍 

The simply the act of logging is very slow may be causing more missed events than without logging.  But I think at this point there's not too much else we can do without alot of rewrite. We're at an acceptable level now I hope.

phantom-8, 2020-12-05:
I've retest the LED issue, it's now fixed.

For test case #4, I first tested without logging.  When problem occurred, I turned on logging (by adding the logging code 'cos it wasn't in the latest source version).  Then, I re-tested again and collected the log, so I can see what exactly was going on.  Therefore, the problem was not resulted from the addition of logging code.

Anyway, I fully agree with you.  The current version is very good and I wouldn't worry about the possibility of checkControllerPresence() causing missing button events.  In fact, I wasn't able to reproduce the problem under normal 2500ms timing.

Again, thank you very much for your excellent work and the help to solving these issues.

Ryzee119, 2020-12-05:
No problem ☺ checkControllerPresence() may even be optional. Could just remove it too

[31 more replies]