Ryzee119/ogx360 #42: Question about modifications to XBOXONE.cpp

jimnarey, 2020-11-24:
Hi - much like Phantom8 (see earlier issue you replied to) I have been trying to build a simplified version of the adapter for one wired controller at a time. I have a relatively quick question - why is rumble disabled in the modified version of XBOXONE.cpp included as part of the USB Host Library?

Thanks.

Ryzee119, 2020-11-24:
I recall it was an issue with wired xbone rumble drawing too much power and causing ogx360 to reboot

jimnarey, 2020-11-24:
That makes sense, thank you.

jimnarey, 2020-11-28:
I now understand the code a bit better. The modifications to XBOXONE.cpp (seem to?) just prevent a rumble taking place on init. However, you were right about the power - a longish rumble was enough to hang the device so in my fork of your project I've disabled it via settings.h.

Thanks again!

closed 2020-12-01 by jimnarey