Ryzee119/ogx360 #44: Adding PS4 controller support

jimnarey, 2020-11-28:
Hi,

Thanks for the advice on the rumble issue. I was hoping you might be able to give me some pointers on extending the project?

I have been working on a fork of the project, with the aim of making the hardware build easier and adding support for more controllers. I've removed things like Steel Battalion support and support for multiple controllers (to help me understand what's going on, as much as anything) and it all works fine. I'm using VSCode and the CMake tools extension.

I am now trying to add PS4 controller support:

https://github.com/jimnarey/SimpleXboxControllerAdapter/tree/add_PS4_controller

I started by copying all the files from the main USB Host Library repo which weren't modified in ogx360 into '../lib/USB Host Shield'.

If I add this to main.cpp:

#include <PS4USB.h>

...it builds fine.

If I then add:

PS4USB PS4Wired(&UsbHost); 

I get the following error from the linker:

Scanning dependencies of target sxbca.elf
[ 97%] Building CXX object CMakeFiles/sxbca.elf.dir/main.cpp.obj
[ 98%] Linking CXX executable sxbca.elf
CMakeFiles/sxbca.elf.dir/main.cpp.obj:(.rodata._ZTV9PS4Parser[vtable for PS4Parser]+0x4): undefined reference to `__cxa_pure_virtual'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/sxbca.elf.dir/build.make:103: sxbca.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:84: CMakeFiles/sxbca.elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

If I completely replace the files in '../lib/USB Host Shield' with the all of the files from the main repo, I get exactly the same error.

If I create a new PlatformIO project to test the USB Host library with just:

#include <Arduino.h>
#include <PS4USB.h>

void setup() {
  USB UsbHost;
  PS4USB PS4Wired(&UsbHost);
}

void loop() {
  // put your main code here, to run repeatedly:
}

...it builds fine!

It seems like there's a virtual method which isn't then impemented in a derived class, but this doesn't happen in PlatformIO. I'm not quite sure where to start? Any help would be greatly appreciated!

Thanks, Jim.

jimnarey, 2020-11-30:
Hi - I get similar errors, e.g.

libUSB_Host_Shield.a(Wii.cpp.obj):(.rodata._ZTV16BluetoothService[vtable for BluetoothService]+0x4): undefined reference to `__cxa_pure_virtual'

...if I try to instantiate the PS4BT or WII classes. I can instantiate the general hiduniversal class. I now have the PS3 controller completely working.

Ryzee119, 2020-11-30:
The __cxa_pure_virtual function is defined in `abi.cpp` as part of the arduinocore. This file has not been copied over to this project.
Hopefully copying this https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/abi.cpp into the `Firmware/ogx360_32u4/lib/Arduino/` folder should do the business :)

jimnarey, 2020-11-30:
Hi - thank you so much for the help. I had thought `__cxa_pure_virtual' was a name produced by the compiler and didn't realise I could have searched for it within the Arduino library. So I'll know for next time.

I tried copying abi.cpp but got errors from the compiler this time, about the contents of the formerly missing function:

https://github.com/jimnarey/SimpleXboxControllerAdapter/blob/test_lib_imports/add_abi_dot_cpp_error

..so I tried copying all of the files in ArduinoCore/cores/arduino into the project and got a different set of errors compiling USBCore, seemingly before it gets to abi.cpp ("‘USB_VID’ was not declared in this scope"):

https://github.com/jimnarey/SimpleXboxControllerAdapter/blob/test_lib_imports/add_all_missing_Arduino_files_error

If I delete the contents of '../lib/Arduino' in the project altogether, copy everything from 'ArduinoCore/cores/arduino' in the Arduino library into it and then replace the missing pins_arduino.h from the Arduino library's 'variants' folder, I get the second set of errors again.

Any more help with this would be really appreciated - I'm really keen to get some more controllers working! Thank you!

jimnarey, 2020-12-01:
Hi - I've sorted it, albeit not in a very clean way. The first error thrown when only abi.cpp is put in the Arduino folder is:

/home/jimnarey/projects/SimpleXboxControllerAdapter/Firmware/lib/Arduino/abi.cpp:25:3: error: expected unqualified-id before ‘[’ token
   [[gnu::weak, noreturn]] void terminate() {
   ^

So I tried removing "[[gnu::weak, noreturn]]" and it compiled. I get a couple of compiler warnings about no return functions returning, but everything works fine and I have just added in PS4 support via USB.

Thank you so much for your help and all the work that went into ogx360!

closed 2020-12-01 by jimnarey