Ryzee119/ogx360 #47: Help: "MAX3421 Revision invalid" in debug program

itsmattkc, 2020-12-08:
Sorry if this isn't the right place for this, I wasn't sure where else to post it.

So I've been trying to build one of these myself and keep running into the same issue printed by the debug program:

```
MAX3421 Revision invalid. Value returned: 00
Unrecoverable error - test halted!!
0x55 pattern is transmitted via SPI
Press RESET to restart test
```

I've built two of these so far and both show this error. My first build actually returns `FF` instead of `00` if that's relevant, but the rest is identical. I'm also using the same Arduino between the two at the moment (they're socketed).

I assumed I must not have soldered the MAX3421 correctly, but through numerous attempts of reflowing all the pins, I've never been able to get past this message.

Is there something else I should be checking or is the MAX3421 soldering job the only real culprit?

chaddoncooper, 2020-12-08:
Would check the soldering on the voltage regulator and that the chip is getting power.

Ryzee119, 2020-12-08:
That SMD crystal is often an issue and can cause the symptom you're seeing.
As mentioned, check all power rails. Make sure that buffer under the 1st arduino is getting power etc.

itsmattkc, 2020-12-09:
Hmm still much the same. I've reflowed pretty much everything (including the crystal), I've confirmed with a multimeter that the regulator is producing 3.3V, and the USB controller and buffer are both getting that 3.3V at their VCC pin.

Only thing I haven't tested is another Arduino, but feels like that's next on the list?

Ryzee119, 2020-12-11:
What happens when you use the standard ogx360 firmware? Does the led flash?

Ryzee119, 2020-12-11:
That being said, invalid revision seems like an issue on the SPI bus.
If you have access to a oscilloscope check the MISO, MOSI, SCK and CS lines on the arduino to see if you get traffic on all of them.
Also try this firmware: It significantly slows down the SPI speed to eliminate SPI speed being an issue
https://github.com/Ryzee119/ogx360/files/5664868/MASTER_360WLESS_360W_STEELBATTALION.zip

itsmattkc, 2020-12-11:
Yes, the LED flashes in normal mode.

I'm afraid I don't have an oscilloscope (probably something worth getting one day though)

I'll try that firmware now and see what happens.

Ryzee119, 2020-12-11:
What 12mhz crystal are you using? From digikey or somewhere else?

itsmattkc, 2020-12-12:
So funnily enough it's not flashing anymore. Same error but no flash, I'm not sure if that's good or bad.

Yep same crystal from digikey listed in the files: https://www.digikey.com.au/product-detail/en/ECS-120-12-30B-AGM-TR/XC2676CT-ND

Slow firmware doesn't make a difference unfortunately.

Still waiting for a second Arduino to arrive so I can test that.

itsmattkc, 2020-12-15:
Unfortunately same story with a second Arduino.

Does anything look wrong on my board?
[image]

Ryzee119, 2020-12-15:
Must be SPI comms issue or MAX chip not booting or held in reset or something.
Check these points
[image]

itsmattkc, 2020-12-15:
Ahhhh I finally got it. Such a silly mistake. A few of the buffer pins indeed weren't soldered correctly. Despite thinking I'd reflowed the whole thing a few times, I guess the solder just didn't get in there (the socket header made accessing it tricky which definitely didn't help).

I'm so sorry to have wasted your time. It works great now, thank you so much for the help and for making such a cool project.

closed 2020-12-15 by itsmattkc