Ryzee119/ogx360 #50: Populated boards

AireagoirSteam, 2021-01-01:
More of a suggestion really.
I was wondering if Gerber files could be added for populated boards or isnt that possable?

Ryzee119, 2021-01-01:
Im not sure I understand what you mean.
Gerbers are a format that describes the placement of copper, holes, silk screen and component footprints etc.

Populated boards must be made in conjunction with the materials list or working with a pcb fabricator to solder the parts for you.

AireagoirSteam, 2021-01-01:
Thank you for the fast reply I remember another project on here a cartridge reader had a way of ordering populated PCB thought it was to do with the Gerber files.

Which I'd why I suggested it.

Ryzee119, 2021-01-01:
I have a new branch that has reworked the PCB and will use kitspace as a ordering system.
Its much simpler just a few clicks to order the boards and components

This will still need self assembly though. 

If you want pre assembled boards theres a few sellers out there.

Here for example 
https://www.facebook.com/groups/TheMODShopByTEK/?ref=share

closed 2021-01-01 by Ryzee119