Ryzee119/ogx360 #51: Soft and hard presses aren't registering in games using 8bitdo and Xbox One Controller Model 1708.

regulardude400, 2021-01-03:
Hi not sure if this is controller related but I have 3 Xbox One controllers that are all Model 1708.  I am using the following setup with my Xbox. 4 Xbox One Controller (Model 1708) -> 4 8BitDo Red Bluetooth Adapters -> Powered USB Hub -> OGX360 -> Xbox.

When playing games like Dead or Alive 3, you can see which input are you pressing. I discovered that with the original xbox controller model s, it registers soft presses and hard presses as two distinct button press types.  When I plug in any of my Xbox One Controller Model 1708, it can only register the controller as soft input only or hard input only. What causes the device to register one over the other is still unknown to me. But it seems that OGX360 is not recognizing soft input and hard input when it comes to games like Dead or Alive 3.  I have linked a youtube video to this to hopefully better explain the problem. Excuse the dust and distracting noises that you may see or hear in the video

Ryzee119, 2021-01-03:
Xbox one controllers do not have pressure sensitive buttons like the original controller.

This is a known issue. The buttons are either fully on or fully off.

Why it sometimes registers soft presses im not sure. Id expect hard presses all the time

regulardude400, 2021-01-04:
Hey just want to say thanks for all of the hardwork you do and that I appreciate your response on this known issue. I didn't look at all of the source code yet, but I guess the buttons are enums. If this is true, would it be feasible to add a button combo for ogx360, like for example: Press select + L + B to switch to soft input mode and hit that same combo again to switch to hard input. I'm not sure how the original xbox even reads soft and hard presses. But if my premise is true in any way, it may be a simple hotfix to work around the issue. Thanks again for your time.

Ryzee119, 2021-01-04:
I should be able to do that no problem. Ill look onto it as soon as i can

regulardude400, 2021-01-09:
I think it is okay to close this issue if you wish Ryzee.  A workaround for this is to disable analog button in the options menu if the game supports this particular setting.  I'm not sure if DOA 2 Ultimate and DOA 3 are the only games to use analog input in this manner but should help others get around the issue hopefully. Thanks again for your time.
[image]

Ryzee119, 2021-01-09:
oh awesome. Glad there's an option.
Ill close the issue but for reference this section of code sets 0xFF(max) or 0x00(min) for the analog buttons. Changing the 0xFF will change the button press intensity. a modifier button or timer thing could be implemented if anyone requires this feature.

https://github.com/Ryzee119/ogx360/blob/73a1a9be35e1146757e6c1803a8bf36d2acedd24/Firmware/src/main.cpp#L249-L255

closed 2021-01-09 by Ryzee119