Ryzee119/ogx360 #52: Button Remaping?

Nashismo, 2021-01-08:
Hi Dear Ryzee!

I bought from you one of these adapters! One of the frist prototypes! I am the chilean guy from the OGXbox forum :) 

I wanted to ask you, maybe in the latest firmware you already added button remaping? Mostly what I would love to have is the ability to remap the right analogue joystick! Mostly for games like Bad Fur Day and Ninja Gaiden Black, that have inverted cameras, which sucks because games today have us used to non inverted cameras.

Do you have something like that implemented already? Sorry to bother you with such a thing!

Thank you so much,
Nashismo

Ryzee119, 2021-01-09:
See https://github.com/Ryzee119/ogx360/releases/tag/V1.61
You can now toggle the Yaxis on the right stick by holding the Right stick in then pressing BACK..

Please let me know how this goes :)

Nashismo, 2021-01-10:
I cannot believe this! You really did this? You are amazing, I will be checking tomorrow, because tomorrow I will have some free time. I will be telling you if everything works fine.

You are a sun Ryzee! :)

Nashismo, 2021-01-12:
> 
> 
> See https://github.com/Ryzee119/ogx360/releases/tag/V1.61
> You can now toggle the Yaxis on the right stick by holding the Right stick in then pressing BACK..
> 
> Please let me know how this goes :)

Dear Ryzee, after some trial en error I finally flashed the firmware correctly, and I got my wirleless 360 controller working great. Sadly, the new feature does not work for me. I flashed the 1.61 firmware but to no avail.

At least in Ninja Gaiden Black it is not working, I press the right joystick stick in, then back, but nothing happens. I have both an original wireless 360 controller and an original microsoft USB wireless receiver.

At least I learned how to update the firmware!

Thank you still Ryzee, maybe when you have time you can take a look at this. Thank again for everything,

Nashismo

Nashismo, 2021-01-12:
NOOOOOO I checked again and sure!! It works! But it is the horizontal axis which needs fixing! The Y axis changes but that is not the problem, it is the X axis the one that is backwards in many games dear Ryzee!

My best regards Ryzee! I hope you can understand what I mean ;)

Ryzee119, 2021-01-13:
Ah right my bad
If you can compile yourself its as easy as changing this section. Replace rightStickY and rightStickY with X.
https://github.com/Ryzee119/ogx360/blob/3c8cdbdc67dfb97554b2dc584dbefda3d08c103b/Firmware/src/main.cpp#L267-L272

Nashismo, 2021-01-13:
Dear Ryzee, I did it exactly as you told me to:
     XboxOGDuke[i].rightStickX = -getAnalogHat(RightHatX, i) - 1; 
 else 
     XboxOGDuke[i].rightStickX = getAnalogHat(RightHatX, i);

 it compiled fine and it works as intended...

But now the Y axis is not working at all! :( I cannot look up and down, etc. 

Forgive me Ryzee for bothering you. I didn't change anything else in the code, only those 4 instances where I replaced Y for X.

Cheers and thank you so much for your time Ryzee

Nashismo, 2021-01-13:
I fixed it myself! There was a line missing in the code:

//Read Control Sticks (16bit signed short)
                    XboxOGDuke[i].leftStickX = getAnalogHat(LeftHatX, i);
                    XboxOGDuke[i].leftStickY = getAnalogHat(LeftHatY, i);
                    XboxOGDuke[i].rightStickX = getAnalogHat(RightHatX, i);
                    XboxOGDuke[i].rightStickY = getAnalogHat(RightHatY, i);     (this one)

After I changed that and recompiled the firmware, everything worked as intended!! Thank you Ryzee

Ryzee119, 2021-01-14:
ah yep youre right, I we needed to add that time back in!
Glad you got it going!

closed 2021-01-14 by Ryzee119