Ryzee119/ogx360 #55: V1.61 Y axis toggle mode does not function

jfoshee900lbs, 2021-02-14:
I just flashed the latest version through the VS code method to try the Y axis toggle mode. For me the right stick just gets disabled and no longer responds.

I am using an 8bitdo wireless receiver on firmware v2.0 hooked directly into a full OG x360 kit with only the master arduino hooked to the p1 port of the xbox. I have tried with both an Xbox one controller, and a PS5 controller, both just cause the right stick to no longer function when the mode is toggled.

The rest of the controller operates as intended, only the right stick is affected.

Also I'll use this moment for a feature request for a hotkey to flip the X axis on the right stick as some games also get that wrong and don't have an option.

Thanks!!

jfoshee900lbs, 2021-02-14:
Goodness I'm sorry, it takes a full minute to make the switch. completely works as intended. My mistake

closed 2021-02-14 by jfoshee900lbs

Ryzee119, 2021-02-14:
It should switch instantly so thats interesting its taking so long.

X-axis isnt implemented at the moment. I can look into it though