Ryzee119/ogx360 #57: Xbox One Wireless Adapter for Windows 10 (model 1790)

mmillar-bolis, 2021-03-04:
Greetings!

I happen to have one of [these adapters](https://www.xbox.com/en-US/accessories/adapters/wireless-adapter-windows) from Microsoft that I would love to donate to you. I would be happy to pay for shipping as well. Of course your time is your own, but do you have any interest in introducing wireless support for this adapter?

Granted, I don't know whether there are any complications in adding yet another adapter onto the OGX code, but I thought that getting the device in your hands may give it a better purpose than just sitting here on my shelf. It's so compact that I would assume others would be interested as well.

Anyway, let me know and feel free to say no! :D

turntablez123, 2021-03-11:
I would be interested in this as well!  I would donate for the ability to have the Xbox one dongle work on the ogxbox!

Ryzee119, 2021-03-17:
Ive looked into this a bit already. Ref https://github.com/Ryzee119/ogx360/issues/6

At this point in time the atmega32u4 chips used on the ogx360 are simply not capable of supporting this alongside everything it already does unfortuntaely.

closed 2021-03-17 by Ryzee119