Ryzee119/ogx360 #59: Unable to Compile Firmware

Quickplay-nl, 2021-03-07:
Hello,

Fist of all great project i made all soldering but i'm stuck on the firmware thing i downloaded Arduino IDE, and also VSCode.

The next step i have to do i don't understand the Setup Visual Studio Code as per the Compiling instructions and cant find out to get into the project environment from the Platform IO toolbar. SLAVE/MASTER etc. 

Sorry for being a big noob but i realy need some help

Hope i asked this at the right place im very new to github

Thnx in advance

Ryzee119, 2021-03-07:
You also need to install PlatformIO plugin for VScode.

The go File>OpenFolder>ogx360/Firmware

Thr toolbar is along the bottom. I think it says 'Default' normally. This can be changed to Master, slave etc

Quickplay-nl, 2021-03-08:
Hi Ryzee119,

Thnx will install it tonight hopefully i get it to work. The hardware is working great used your debug hex te test it.

If i run into other problems can i ask for some help if needed?

Regards Ferry

Ryzee119, 2021-03-09:
Hi yes, I'll close this issue for now, but you can comment here or if its a new issue raise an new one.

If you have worked it out and feel the instructions could be improved youre welcome to raise a PR to improve it :)

closed 2021-03-09 by Ryzee119

Quickplay-nl, 2021-03-10:
Thnx for your help I manage to get the correct HEX files with your suggestions. I hope to receive my xbox 360 receiver this week so I can test everything. 

Thnx for your great project and again for the support. 

Regards Ferry