Ryzee119/ogx360 #69: Is there a CPL file to have them ordered pre populated with the smt parts?

ChuxtyGT, 2021-06-25:
Sorry if this isn't allowed.  I just want to build a few for the 3 xboxes I have and wanted to know if they can be ordered with everything already mounted on the pcb except of course the arduinos.  I am capable of soldering the parts but was wondering if that's possible?  Unless I missed the CPL file somewhere?  Thanks for any info.  This is such a great project!

Ryzee119, 2021-06-27:
There currently isn't anything like that setup. The easiest way to order at the moment is https://kitspace.org/boards/github.com/ryzee119/ogx360

If someone wants to create such a thing and raise a PR that would be welcome.

SavageCore, 2023-06-08:
@Pouletgrill Be interested in seeing that! Mostly what you chose for J1 and Y1 as the parts aren't on LCSC?