Ryzee119/ogx360 #70: Bill of materials?

Electric1447, 2021-07-06:
I remember being a link to digikey with all parts need for this project listed.
Was it removed or am I just blind?

Electric1447, 2021-07-06:
On a different note I have multiple Mayflash adapters and q ridiculous amount of controllers, I would love to contribute to this amazing project.

Ryzee119, 2021-07-06:
See hardware folder readme. I use kitspace to simplify ordering via digikey/gerbers

https://kitspace.org/boards/github.com/ryzee119/ogx360

Ryzee119, 2021-07-06:
Have you tried the mayflash adaptors? If they're xinput compatible it may just work

Electric1447, 2021-07-06:
They have an xinput mode

Downcraft, 2021-07-13:
Hi, first of all thank you for this awesome project. Do you have an alternative for the max3421 controller? I have a hard time getting one. I tried digikey/aliexpress/amazon and some other sellers it seems to be sold out.

Ryzee119, 2021-07-17:
Max3421 is critical to providing the usb host functionality and all the code is based around that chipset. So this project relies on it and it cannot easily be substituted. 

I suspect the global chip shortage is not helping supply.

closed 2021-07-17 by Ryzee119