Ryzee119/ogx360 #72: Materials

florezsebas, 2021-07-16:
Can I use components of another manufacturer while this components has the same values like volt, capacitance, resistance?
Sorry for my ignorance about this, but I don't know much about electronics

Ryzee119, 2021-07-17:
Yea pretty much.

Most parts are nothing special. The crystal oscillator and the diodes may want extra attention. 

The crystal should have matching load capacitance for example.

Crossfader84, 2021-09-12:
Hi, i have a question to the USB Controller IC.
The MAX3421EEHJ+T is not in stock and comes back early 2022, can i use the MAX3421EEHJ+ ?
Or is there an other optional IC. This is the only Part on my list that is not in stock :-(

Ryzee119, 2021-09-12:
Whats the difference? The T is probably just something to do with packaging. So probably fine

Crossfader84, 2021-09-12:
> Whats the difference? The T is probably just something to do with packaging. So probably fine

Thanks a lot, finally i can order and build the OG360 :-)

closed 2021-12-16 by Ryzee119