Ryzee119/ogx360 #74: Problems compiling.

xboxnutter, 2021-08-17:
Hi,

So I have installed the Arduino IDE (already had for the spi2par2019 project), but checked it was still updated.
Installed Visual Studio Code - no problems.
Installed the extension PlatformIO IDE & let it install it's dependancies etc. -no problems.
Installed Git.
Restarted VSCode.
Added the Github link to clone the repo - seemed to be fine.

Opened up the Firmware folder in VSCode, then hit build, then fails to find 'Arduino.h'

This appears that it should be residing inside the folder src>>lib>ArduinoCore-avr which was empty, so manually added this from the Arduino repo with all the other folders as 'Arduino.h' is inside a subfolder 'Core' within that.

That appeared to work - build then stopped & failed again for the empty UHS2 folder now instead. So copied all the files from that repo into that folder as well.

Now the build works and also programs to the OGX360 fine. Just one underline problem about the intellisense can't fine the 'include' folder. So added empty include folder to Firmware folder & refreshed.
That worked.

After all that, the OGX360 would not function properly and could not get the 8bitdo or anything to work. Spent 2 days faffing about double checking everything hardware wise including the 8bitdo & updated that as well.
No joy.

Ended up then using Avrdude in cmd with the .bat using latest release. Uploaded fine to both modules (2 player) and now functions 100% perfectly.

Tried the VSCode compile many times since, same issue. So built the compile on my macbook pro instead, ran into exactly the same issues as above - had to manually add missing folder files etc - tried full zip file expand and files added again instead, that didn't work either.

Yes I have it working, would just like to get it working via the compile method as I want to curtail the other PC.

Thanks for the top work on this, the spi2par & the openxenium too!

Ryzee119, 2021-08-17:
You need to clone this repo recursively. From the compiling instructions:
`git clone https://github.com/Ryzee119/ogx360.git --recursive`
I suspect its not working because you're manual method is pulling the wrong commits.

Ryzee119, 2021-08-17:
I just manually trigger an auto build via Github Actions and it built fine. Hex files appear identical to the release build.

https://github.com/Ryzee119/ogx360/actions/runs/1141063339 (may need to be logged into github to see log)

xboxnutter, 2021-08-18:
Thanks for the reply.

That's the link I put in the VSCode box after clicking clone repo. That's what I did first. But the build stopped at missing the 'Arduino.h' file.

What does the recursive mean? Sorry, I did bing search that but it gave me a technical definition so have no idea. When I add the git clone link it seemed to work fine, but obviously I'm not doing something correctly?
I added those other files manually afterwards as they were missing.

Complete noob to this :o)

Ryzee119, 2021-08-18:
Recursive means it will clone to main repo, but also get the submodules (which is the AVR library and the USB library)
Not sure what method you are using to clone:

* If you are on windows install git for windows, and in a command terminal type `git clone https://github.com/Ryzee119/ogx360.git --recursive` that should work.

* On Linux, install git  through the package manager and use the same command in a terminal.

xboxnutter, 2021-08-18:
Just tried that and got "unable to access fatal - URL error 400" pop up.

xboxnutter, 2021-08-18:
Validating found git in: /usr/local/bin/git
Using git 2.32.0 from /usr/local/bin/git
> git clone https://github.com/Ryzee119/ogx360.git%20--recursive /Users/sicko/Downloads/OGX360/ogx360.git --recursive --progress
Cloning into '/Users/sicko/Downloads/OGX360/ogx360.git --recursive'...
fatal: unable to access 'https://github.com/Ryzee119/ogx360.git%20--recursive/': The requested URL returned error: 400
> git clone https://github.com/Ryzee119/ogx360.git%20--recursive /Users/sicko/Downloads/OGX360/ogx360.git --recursive --progress
Cloning into '/Users/sicko/Downloads/OGX360/ogx360.git --recursive'...
fatal: unable to access 'https://github.com/Ryzee119/ogx360.git%20--recursive/': The requested URL returned error: 400

doh lol

Ryzee119, 2021-08-18:
Yea you're not cloning it properly 🙃

xboxnutter, 2021-08-18:
Yeah I've tried everything, turning anything that could be blocking it off, but no luck. Dunno, won't do it recursive, just get error 400 all the time.

Ryzee119, 2021-08-18:
git clone --recursive https://github.com/Ryzee119/ogx360.git
Could try this i guess

xboxnutter, 2021-08-18:
Nah, gives 'unknown option' with that one.

xboxnutter, 2021-08-18:
Must be a network issue. On 3's mobile 4G broadband uk. Don't have landline.

Ryzee119, 2021-08-18:
Ill provide a zip later. Im out of office at the moment

xboxnutter, 2021-08-18:
No worries, off to sleep now at 6:18am after sitting here all night lol

Thanks for the help.

xboxnutter, 2021-08-18:
Just noticed the url is changed by VSCore in the error log...

> git clone https://github.com/Ryzee119/ogx360.git%20--recursive /Users/sicko/Downloads/OGX360/ogx360.git --recursive --progress
fatal: unable to access 'https://github.com/Ryzee119/ogx360.git%20--recursive/': The requested URL returned error: 400

Clicking the link by cmd+click shows url is wrong for repo. Odd. Why is the visual studio changing the url?

Ryzee119, 2021-08-18:
[image]

[ogx360-814e1ed.zip](https://github.com/Ryzee119/ogx360/files/7005389/ogx360-814e1ed.zip)

Cloned recursively and zip attached. This should compile is VSCode fine

xboxnutter, 2021-08-18:
Thanks very much for that zip.

After a bit more googling it appears others have this problem with VSCode, so tried various fixes, none worked so far. Apparently they fixed it in one release, then it broke again :o/
Might try SSH instead.

closed 2021-08-18 by xboxnutter

Ryzee119, 2021-08-18:
I dont use git within VSCode so not sure what was going on there. I just use a terminal window with git