Ryzee119/ogx360 #76: debug hex file

UnElectronicoMas, 2021-10-07:
Hi, @Ryzee119 i built a simpified version of ogx360 adapter using the mini USB host shield 2.0 following the issues post called "Customized ogx360 hardware with Pro Micro 3.3V " and now i want to test it, the problem is when i open the firmware on platformio, visual studio only show me the OGX360-master/Firmware environment, how can i compile the debug hex file to test the board? i need to modify somthing on your code to compile the debug version? thanks in advance

Ryzee119, 2021-10-18:
Hi @UnElectronicoMas , apologies for the delay,

The QC test program is basically this: https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/board_qc/board_qc.ino

The only thing you need to add early in the setup() function, before `Usb.init()` i this:
```
#define USB_HOST_RESET_PIN 9
pinMode(USB_HOST_RESET_PIN, OUTPUT);
digitalWrite(USB_HOST_RESET_PIN, LOW);
```

Ryzee119, 2021-12-16:
Hope you got it all working! Please let me know if you found anyhting else

closed 2021-12-16 by Ryzee119