Ryzee119/ogx360 #79: 8bitdo adapter 2

25glitch, 2021-12-14:
Just got 8bitdo adapter 2 stick adapter 1 not responding to Xbox series x controller on ogx360 and 8bitdo adapter 2 won’t stay powered on anyone have a solution to this

Ryzee119, 2021-12-14:
Thanks, im aware of this issue but do not have this new adaptor to test.

Could you please:
1. Provide a usb descriptor dump of the device. On windows I use this https://www.thesycon.de/eng/usb_descriptordumper.shtml
2. Ideally use wireshark to get a enumeration log from Windows which I assume it works in Windows

25glitch, 2021-12-14:
Information for device IDLE (VID=0x2DC8 PID=0x3107):

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x001D
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0110	bcdUSB
0x00	bDeviceClass      
0x00	bDeviceSubClass   
0x00	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x2DC8	idVendor
0x3107	idProduct
0x0200	bcdDevice
0x01	iManufacturer   "8BitDo"
0x02	iProduct        "IDLE"
0x03	iSerialNumber   "E417D89BB8E9"
0x01	bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09	bLength
0x02	bDescriptorType
0x0029	wTotalLength   (41 bytes)
0x01	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0x80	bmAttributes   (Bus-powered Device)
0xFA	bMaxPower      (500 mA)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0110	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x0025	bDescriptorLength

Endpoint Descriptor:
------------------------------

25glitch, 2021-12-14:
Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x08	bInterval         (8 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

--------------------------------
String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "8BitDo"
0x02   0x0409  "IDLE"
0x03   0x0409  "E417D89BB8E9"

------------------------------

Connection path for device: 
USB xHCI Compliant Host Controller
Root Hub
Generic USB Hub
IDLE (VID=0x2DC8 PID=0x3107) Port: 3

Running on: Windows 10 or greater (Build Version 19043)

Brought to you by TDD v2.17.0, Feb 23 2021, 14:04:02

25glitch, 2021-12-14:
Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (1 x 64 bytes)
0x08	bInterval         (8 frames)

Ryzee119, 2021-12-15:
Thanks, I think there's something tricky going on here. Does it eventually appear as a xbox controller in Windows?

`0x02 iProduct "IDLE"` is weird. It probably is looking for some commands to get it going at startup. 

I need to try grab one, but its sold  out by the looks of it

25glitch, 2021-12-15:
it's on amazon got one two days ago

25glitch, 2021-12-15:
it might have something to do with microsoft switching to from Classic to BLE bluetooth last september

Ryzee119, 2021-12-15:
@25glitch glad you got one but theyre sold out now

So this will be on hold until they're back in for me unless someone else investigates it further

CaptMoore, 2021-12-22:
If it matters, they are available on amazon again. I just ordered one for 2 day delivery.  

Thanks for looking into it.

Ryzee119, 2021-12-22:
@CaptMoore thanks for the heads up
Ive ordered but im in Australia so prob a few weeks away

Ryzee119, 2022-01-12:
Ive got my 8bitdo receiver now and can replicate this issue. I'll investigate asap to see if we can get it going

Ryzee119, 2022-01-13:
Progress. Still needs a bit of love, but I worked out what it's doing.
[image]

Ryzee119, 2022-01-15:
[firmware 9aee4cd.zip](https://github.com/Ryzee119/ogx360/files/7874156/firmware.9aee4cd.zip)
Please try this firmware. Depending how old your firmware is (not updated since ~june 2021)  you may need to update all arduino modules.

CaptMoore, 2022-01-15:
Thanks, I am going to go out of town tomorrow for a week, but can check it after I get back.  I am sure others will be able to check it first.

Ryzee119, 2022-01-15:
From what I can tell, if the blue LED on the 8bitdo adaptor blinks twice every second or so, its initialised ok and looking for a previously paired controller.
If the LED stops blinking, ive done something wrong.

Just pair a controller as per normal and once a controller is paired and connected the LED goes solid and we're in business.

Ryzee119, 2022-02-22:
Closing. This has been merged into master branch. and got feedback from a couple people that this is working.

CaptMoore, 2022-02-22:
Thanks. Life has gotten in the way of testing this myself, and the task of flashing the ogx via command lines is pretty daunting (but I have done it before) so I never tried to run the beta firmware. I am looking forward to updating this when life lets me.

senolfeldmann, 2022-06-11:
This issue is present on firmware 2.15. The 8bitdo adapter 2 shuts itself off or does not receive power after ~1-2 seconds when plugged in directly into the ogx360s USB-A socket (only blinks shortly two times). The red led of the master arduino does not light up, too.
The behavior should be like with the 8bitdo adapter 1: The led of the adapter blinks shortly two times, pause, blinks shortly two times, pause, etc... until you press the sync button or connect an already synced controller. Additionally, the red light of the master arduino indicating a connected device lights up.
Strangely, the 8btido adapter 2 works as it should when I do not plug it in directly, but use a usb 2.0 extension cord in between.

Ryzee119, 2022-06-11:
I may need to revisit to see something has changed 🤔

Ryzee119, 2022-07-12:
Looks like

Ryzee119, 2022-07-12:
Looks like the FW update to 1.1 broke something. Feedback is that it still works on 1.0 FW which is easy to downgrade to.
Ive tried to fix it but not had much luck yet

25glitch, 2022-07-12:
I’m thinking the board may have to be rebuilt from scratch to accommodate the power requirements for the adapter 2

Dempsey86, 2022-07-17:
> Looks like the 8bitdo FW update to 1.1 broke something. Feedback is that it still works on the 8bitdo 1.0 FW which is easy to downgrade to.
> Ive tried to fix it but not had much luck yet

I do confirm this, now they ship new with v1.01 and there is a update to v1.02 (doesn't work). Reverted the FW to v1.0 and the stick works.

dleer, 2023-04-15:
I bought the 8bitdo adapter 2 last week and also just got an ogx360 mini. They seem to work perfectly together. I played some games for a few hours without issue using a 2020 model Xbox One controller.

There is one little quirk in that the 8bitdo won't automatically reconnect to the gamepad on power up of the Xbox. I have to press the 8bitdo's power button and then it connects. On a Windows PC the adapter does automatically connect.  I updated the firmware of the 8bitdo to the most recent but the behavior with the ogx360 is the same.