Ryzee119/ogx360 #80: Firmware updater GUI

Ryzee119, 2021-12-16:
Alot of users have issues updating the device. Its not too hard, but for a non technical user its probably not ideal.

We could have a GUI that prompts the user what to do and update the firmware.

CaptMoore, 2021-12-22:
I am not terrible with technology, but a simple firmware update app would be much appreciated.  I accomplished the update, but I had to try it a few times until I did it correctly. 

If it’s not too much to ask, after a windows app is developed could there also be a macOS & Linux version?

Ryzee119, 2022-01-15:
`If it’s not too much to ask, after a windows app is developed could there also be a macOS & Linux version?`
Yes this would be ideal. Something Python based may be the easiest

dtomcat, 2022-07-18:
I just seen this... and I may be able to help...    No promises (GUI is not my cup of tea), but I'll attempt this and hopefully get it functional, then someone else can make a "pretty" GUI.   again no promises, but I'll look into it after I finish the automated OpenXenium programmer.

dtomcat, 2022-07-18:
and sorry... I would only be able to Windows... I'm no bueno with MAC/Linux

dtomcat, 2022-07-18:
[image]
 Progress!!..  Got a lot of it done... still needs work... but I say it went from 15% chance of being done... to 98%!

dtomcat, 2022-07-19:
Doing a lot of error checking (i.e. more than one pro micro connected... more than one pro micro put into programming mode at same time, etc).   Do you think we should add checksums to ensure user doesn't try to upload a correct hex file?  This would add a little more to your plate... with each firmware update, you'd need to add current build to file of checksums to validate against?  I'm going to make this as user friendly and as hard to screw up as possible.  I'm even checking for legit pro micros vs clones (still allowing to upload firmware, but in case an issue arises in future, we can tell if it's only clones, etc).

Ryzee119, 2022-07-19:
I wouldn't worry about checksum checking. Avrdude will verify the flash normally and as you say it would be a bit tricky to manage all the different checksums

dtomcat, 2022-07-19:
[Sneak Peak!](https://www.youtube.com/watch?v=kPFs9sGHJw8)

dtomcat, 2022-07-21:
@Ryzee119 mind testing first Alpha?  basically as seen in sneak peak...  this mostly just check that it will work on a PC that isn't mine
[OGX360 Firmware Updater.zip](https://github.com/Ryzee119/ogx360/files/9155769/OGX360.Firmware.Updater.zip)
.

Ryzee119, 2022-07-21:
Yep can do in the next day or so. Good work so far and definitely will help alot of users. Do you have source code on github anywhere?

Do you plan to opensource this and what license?

dtomcat, 2022-07-21:
Not yet.  It’s a bunch of ideas thrown together with no rhyme or reason.  Once I get a good beta… I’ll clean up the code and document and post to GitHub.   Will definitely be open sourced and have no clue on license.  Not familiar with them. I usually just post the code. Maybe someone can help me with which license I should do and how?

dtomcat, 2022-07-21:
Ok... a working initial Alpha version:
[OGX360 Firmware Updater.zip](https://github.com/Ryzee119/ogx360/files/9157520/OGX360.Firmware.Updater.zip)

Fingers Crossed!

Ryzee119, 2022-07-21:
I tried it out, and am hitting avrdude.exe: ser_open(): can't open device "\\.\COM5": Access is denied, but weirdly it programs find if I just run avrdude from the commandline separately.

I have a couple suggestions for your consideration

* Make the text "Firmware:" a hyperlink to the github release page
* Make "avrdude.exe" a hyperlink to the avrdude download page, also libusb0.dll
* I initially didnt have libusb0.dll in my folder which altered me perfectly, however after adding it in I had to close and reopen the program for it to find it. Maybe a rescan button or something could be useful.
* After clicking "Ready?" This window popped up as expected however the write button was hidden and I had to expand the window to see it. I had to watch your video above to work it out. Hopefully this can be fixed.
[image]

EDIT: It just started working :)
[image]

dtomcat, 2022-07-21:
Thank you for the feedback!  I believe the com5 issue at first would have been that the Pro micro timed out.  i've noticed that you only have a few seconds to initiate an upload before the pro micro resets out of programming mode.  (I'll add a note about this in the text).  I tried to create hyperlinks to no avail (i'll continue trying).  Did you get a splash screen initially telling you the requirements?  I had an issue after adding an icon... so I'm wondering is the issue wasn't from the icon ... I had a label disappear...  like you had the write button hidden...   not sure why that is... I'll try to reproduce that.  A rescan mechanism is definitely in the works!   that is weird... not sure why it didn't expand all the way... as you see in the video, it does on mine.  something I'll have to look into as well.  Thanks again for the feedback... long road ahead... but mostly cosmetic!  Main purpose of app is funtional

dtomcat, 2022-07-21:
[image]
[image]
Solutions....   Thoughts?

dtomcat, 2022-07-21:
@Ryzee119     another build  (has a debug function)... when the window resizes, and you have to manual make it bigger...   a windows will pop up with width/height ... can you send me those numbers?
1. barely resize it (like one pixel)  copy numbers
2. Resize it to make it look like video and copy those numbers

Thanks!  New features above are also incorporated
[OGX360 Firmware Updater.zip](https://github.com/Ryzee119/ogx360/files/9158488/OGX360.Firmware.Updater.zip)

dtomcat, 2022-08-26:
Just programmed 25 chips easily, zero errors, and quickly!  I think the functionality is spot on and ready… just need to work on the cosmetics. 
[image]

dtomcat, 2022-09-25:
Just tried the program on my ogx360i-4ch boards I made... worked perfectly!  I'm confident that all that needs addressing is cosmetic issues. (at least on my PC).