Ryzee119/ogx360 #81: ogx360 can't start Steel Battalion

hpx1984, 2022-02-13:
Dear Sir Ryzee119

i'm used USB Host board and 3.3v/8M pro-micro board(first time i use 5v/16M but it can't working) for build a ogx360 and it can worked for xbox360 pc wireless receiver and 8bitdo wireless receiver 
the FW is use this one
[image]

now  i can't start Game of Steel Battalion, is it means this FW can not support the key map of Steel battalion, 
how i can fix my ogx360 board for support Steel battalion
[image]
[image]

by the way in your OGX360 Project's Pro micro board is 5v/16Mhz ,right?

many thanks your support sir

Ryzee119, 2022-02-13:
Get the latest fw from this github page.
Edit platformio.ini and change the 16mhz numbers to 8mhz and recompile and program

Ryzee119, 2022-02-23:
Did you have any luck with the latest firmware?

hpx1984, 2022-02-24:
Dear Sir Ryzee119

i had download the latest firmware in your project and i know how to change 16mhz to 8mhz in ini file 
and i also downloaded and installed Arduino tool in my PC but i don't know how to recompile to HEX file 
I may study how to use Arduino later 
or could you help for recompile a 8mhz HEX?

hpx1984, 2022-03-13:
Dear Sir Ryzee119

i followed this guide try for compile a 8Mhz FW (i compile 16Mhz original FW first )
[image]

but got many error reported 
[image]
[image]
[image]

i will try to study on Visual Studio Code usage hope can know how to fix the problem

hpx1984, 2022-03-13:
[image]

hpx1984, 2022-03-28:
Dear Sir Ryzee119

could you add one more FW version of 8Mhz in your next FW update release  
[image]

now this FW can support PS3 controller so i think it also has possible for support PS3 Move controller for light gun play with House of the Dead , and I think the FW also is big challenge

Ryzee119, 2022-07-12:
8Mhz wont be officially support by me as I am unable to test it properly and cant comment on the impact of it. 

Not finding Arduino.h header is not related and likely due to not cloning the repo properly.

closed 2022-07-12 by Ryzee119