Ryzee119/ogx360 #85: user issue flashing firmware

Talamis, 2022-04-18:
it flashes the firmware onto the new pro-micros with different brands
but it doesnt want to be recognized by the controller/xbox

the old one i flashed quite a while ago works on the new board flawless

Ryzee119, 2022-07-12:
Any update on this?  Could you provide more info like what brands etc.

wiredopposite, 2022-08-25:
I think I've had this issue as well, I managed to brick an Arduino I was using as a test unit. I had the debug hex programmed to it, then I programmed it again with Avrdude with the normal v2.15 fw (might have been the problem.) The wireless adapter would turn off when I tried to sync it, so I used Avrdude to flash back to debug but when I tried to use Termite, it switched modes/COM number on its own and can't be seen. VS code also won't program it, it'll start but when but I get to the point where you need to press reset, it doesn't do anything and just fails. I tried uninstalling in Device Manager and plugging it back in but after that it didn't even get detected anymore unless it was in bootloader mode.

Ryzee119, 2022-08-26:
TZT ones are normally fine. Ogx360 has to disable the standard arduino usb interface which is used for programming  for it to work. 

The button on the board has to be used reload into the bootloader but it can be a bit tricky to time.

Sometimes you have to double press or hold and release when its about to program.
But I mean it's not possible for just this software to 'brick' it completely.