Ryzee119/ogx360 #86: ogx360 doesn't working ,need help

hpx1984, 2022-05-08:
Dear Sir

i had build 2sets ogx360 but all can't working 
i used pro micro board 5v/16Mhz, test 2.0 and 2.1 FW , got same issue
when i plug ogx360 into xbox the green led turn on , i plug xbox360 controller or 8bitdo receiver into ogx360 
the xbox360 controller LED flash two times and then turn off, 360 controller doesn't working 

i checked the soldering workmanship and i didn't find any defect 
USB vcc is 4.8v
can any one help for comment where i can check or verify 
please help me 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Ryzee119, 2022-05-08:
What value are the caps on the crystal? C2 and c3?

Alot of components look like substitutes so not sure if they are all suitable

hpx1984, 2022-05-08:
Hi sir

it is 16p , but i Measured it actual it is 15p

i have some 0603 size's cap and resistor on hand ,the value is match the bom request so i use then
i try to buy 0805 component for replace these 0603 components

hpx1984, 2022-05-13:
Hi Sir

now i used 0805 chip component for builded 3sets ogx360 still got same issue , the pro micro board just green LED on when i plug xbox360 wire controller into USB connector and the controller just flash two times LED on and then LED off
[image]

[image]

some times red led will flash on pro micro board , dis-connect 360 controller also keep red led flash
[image]

Material status
[image]
[image]
[image]
[image]
NXP 74HC4050D
MAX3421EE
49SMD-12M
10uf smd e-cap

if you have any idea please comment and help me for fix 
many thanks Sir

hpx1984, 2022-05-13:
missed one more component is MIC5219-3.3BM5(marking:LG33)

Ryzee119, 2022-06-06:
The two lights on solid would indicate that is it working to me? Any news on this?

hpx1984, 2022-06-18:
now my ogx360 can worked now ,after i use another crystal oscillator 
but i got another problem 
i use orange 8bitdo adaptor for pairing one PS4 controller and it can worked fine , then i add one more Arduino board for support 2nd controller but i can't pairing the 2nd PS4 controller , tried 2.1 and 2.15 FW and got same issue

hpx1984, 2022-06-18:
360 controller can pairing 2sets, very thing fine , thanks your great project Sir

closed 2022-06-18 by hpx1984