Ryzee119/ogx360 #93: OGXbox Controller Support

XeniumMods, 2022-10-02:
I'm not sure that is anyone is aware but I can confirm that the
OGX360 is compatible with wired Original Xbox Controllers.

I faintly remember suggesting backward compatibility to Ryan
back in early 2020 as this would make internal variants much more
practical but it either wasn't possible or there wasn't enough room.
Even though I've had working OGX360's this entire time, it wasn't
until I did some thorough testing this afternoon that I realised that
OGXbox controller do in fact work with the OGX360 when paired with
a OGXbox to USB adapter cable. I dunno if I missed something during
one of the firmware updates or if this was completely unintentional
but I don't see it as being a supported controller in the overview.

I hope this helps, Nem

Ryzee119, 2022-10-02:
Added in v2 😊

XeniumMods, 2022-10-02:
Oh, right...

Just going over the release notes. It's not very apparent but I assume it was apart of the driver update?

closed 2023-04-07 by Ryzee119