Ryzee119/ogx360 #94: Wireless Xbox Controllers vis USB Cable

XeniumMods, 2022-10-02:
While I was doing some testing on the OGX360, I remembered
there were cables for the Xbox360 so that you could connect
a wireless controllers to the console, possibly for rechargeable
batteries. Not having one to test I thought if something similar
could be done with XboxONE and Series X/S wireless controllers.
Using an Elite 1 (1698) and Series X (1914) controllers, I connected
them to the OGX360 via USB cables I found that they both did
register some button presses but for the most part it would not
work as you might expect.

Is this something that can be implemented?

Ryzee119, 2022-10-02:
Pretty sure wireless 360 usb cables are purely charging?

Can you check if it works on a windows PC?

XeniumMods, 2022-10-02:
That's what I thought. [Xbox Support: Connect Controller to PC](https://support.xbox.com/en-AU/help/hardware-network/controller/connect-xbox-wireless-controller-to-pc)
I got them both to work on PC and tested them over at [gamepad-tester.com](https://gamepad-tester.com/)
But I had to use different cables for them in order for Windows to detect them properly.
Came across a [YouTube video](https://www.youtube.com/watch?v=kzJ2vh58KQ4) that discusses issues with some USB cables.
After this I retested both controllers using the new cables. The XboxONE Elite controller
worked fine and was surprised to see that my presets my the paddles were still there.
The Series X controller does work now but some buttons are not as responsive.
I went back to the PC and tested this controller again to see if it might be the cable
but it seems to be alot more responsive on the PC that it does on the OGX360.

Ryzee119, 2022-10-02:
Are you using the latest fw? V2.15 fixes some issues with xbox s/x controllers

XeniumMods, 2022-10-03:
I just updated my Series X controller using my Series X console,
not sure what version the firmware is but the issues persist.
It could be the cable I'm using but it's the Xbox USB cable that
came with my Elite 2 controller so you would expect it to work.
Unfortunately I don't have much in the way of Type A to Type C
cables to test any further at this stage...

I'm wondering if zeroing out those dropping resistors on the
data lines just behind the Type-A port would help?
I noticed on the tinyogx360 they were removed completely.

Ryzee119, 2022-10-03:
I mean ogx360 firmware v2.15

XeniumMods, 2022-10-03:
Oh right, sorry my mind was else where.
Confirmed that firmware v2.15 did the trick.
Not sure which version I was using...

Thanks again

closed 2023-04-07 by Ryzee119