Ryzee119/ogx360 #99: issues getting 3 parts for the OGX360

Jonnycake99, 2022-10-18:
Having issues getting 3 parts for the OGX360 from digikey and alternative places. 
Can anyone help me get substitutes that wont damage the circuit.

Reference (U2)                 74HC4050DCT-ND 
Reference (D1,D2,D3)      1727-7328-1-ND
Reference (U3)                  576-1281-1-ND

Thanks,

yankz12, 2022-10-26:
Hey Jonny,

I uploaded an item list with current availabilites which applies for Mouser & Digikey in Germany:
[availability_items.ods](https://github.com/Ryzee119/ogx360/files/9870028/availability_items.ods)

Where do you try to order from? Depending on the country, availabilites can differ. If I split the parts in two different orders, I can buy everything except for the Schottky Diodes (BAT54GWJ). I did some research via Digikey and the part which is closest to the BAT54GWJ is the BAT43W-E3-08. Differences are:

- Higher Voltage - Forward: 450 mV @ 15 mA instead of 400 mV @ 10 mA
- Smaller Current - Reverse Leakage: 500 nA @ 25 V instead of 2 µA @ 25 V
- Smaller Capacitance: 7pF @ 1V, 1MHz instead of 10pF @ 1V, 1MHz

By substituting the Diode with the BAT43W-E3-08 ones, are there any issues that could arise? Or is this safe to apply? Sorry, I dont have much of a clue regarding Diodes...

Kind regards,
yankz

bvinson9, 2022-11-28:
I can confirm I have sourced all components for making 4 or 5 complete 4 port OGX360's the only parts I can't get are the BAT54GWJ Diodes. I even ordered 25 of them from Newark.com and was shipped an empty envelope. They took some prodding, but refunded me. Then I reordered and was emailed that my order will be shipped after 07/18/2023.

I was looking at BAT42W-G3-08 on mouser.com. I'm not even close to knowing how to make this circuit work w/o proper supervision. Hoping to get some help here. Please.

locutus2000, 2023-01-30:
Would BAT42W-E3-08 be a suitable replacement? BAT54GJ is still not available (at least not digi-key).

https://www.digikey.nl/en/products/detail/vishay-general-semiconductor-diodes-division/BAT42W-E3-08/3104176?s=N4IgTCBcDaIEYEMAuAWMB3AtAUwMyYAYAOAYyUwDsATEAXQF8g

closed 2023-04-07 by Ryzee119