Ryzee119/ogx360 #100: Parts list seems incorrect.

thalion777, 2022-12-08:
[image]

Not sure which part quantities are accurate, but i feel like 5 reset buttons is wrong...

Ryzee119, 2022-12-08:
Most are rounded up to the nearest multiple of 5 except things that cost more than a few tens of cents

thalion777, 2022-12-08:
Ok gotcha!  Btw, I can't for the life of me figure out what the TP1 - TP4 is.  All the videos I see on youtube has those 4 circles missing.

thalion777, 2022-12-08:
Much appreciated friend.

closed 2022-12-08 by thalion777

SavageCore, 2023-06-08:
> They're just test points for debugging. They are just exposed pads on the PCB that you can put a meter or oscilloscope onto
> 
> It's specifically for debugging the spi bus between the atmega and max3421

Ahn thank you. Should this be removed from the Kitspace interactive assembly guide?