Ryzee119/ogx360 #101: Using black 8bitdo WA2 - won't work in a powered/unpowered hub or 2 brown 8bitdo 2 player.

xboxnutter, 2023-01-15:
Hi Ryzee,

Have come across this issue where the black 8bitdo WA2 works perfectly when it's plugged directly into the OGX360 usb port, also syncs with series x controller and works as it should.

Yet, when the above 8bitdo WA2 is plugged into a 2.0 USB hub (belkin or any brand we have tried) and the hub plugged into the OGX360 - then it does not work at all. The controller will sync with the 8bitdo, but there's no response to the ogx360 and the primary arduino second led does not light either. Tried hub powered and unpowered.
Tried reset on the ogx360 etc, still nothing.

Unplug the hub and fit the WA2 adapter back in and works again.

Now with the brown 8bitdo - this works in the hub, yet as soon as you plug in the black wa2 second 8bitdo, then neither works!

I have tried 3 OGX360's and another lad I message has tried his as well. He has tried 2.1 firmware as well, still the same.

I am using 2.15 firmware as well. It seems there is something wrong, we have tried a few different hubs to no avail.

So to recap;

Black 8bitdo works directly plugged into ogx360, but not in any hub.

Two brown/orange 8bitdo's plugged into the hub, one works but player 2 8bitdo does nothing but syncs to controller.

Also, with the brown/orange 8bitdo working in the hub alone, if you plug in a wired 360 controller then this works for player 2!
Really odd.

Any help appreciated.
Neil.

xboxnutter, 2023-01-20:
Update;

Works with multiple 'Brown' 8bitdo adapters in a 2.0 hub **_only_** if the 8bitdo firmwares are all V2.03 (V2.04 breaks this).

Only the original earlier 8bitdo works - the V2 8bitdo (with a 2 on the body) and the black V2 8bitdo will not work in the hub. No matter what firmware. They only work plugged directly into the OGX360 main USB port with no hub. (single)

So appears the OGX360 firmware would need upgrading to cope with both the latest V2.04 8bitdo firmware for original brown dongles and also entirely for V2 dongles (new type) to be used in a USB hub for more than one player.

So for multiple 8bitdo dongles in a USB 2.0 hub they need to be downgraded or left on V2.03 firmware (older brown type) - grey might work on v2.03 also (untested).

wiredopposite, 2023-05-23:
After some testing, I think this is more of a USB hub/8bitdo issue. None of the cheapo 2.0 hubs I've tried have worked with the 8bitdo v2 (sl2.1a and another popluar Chinese IC, don't remember the name), both through the OGX360 and on PC. Controllers sync but no input is read. 

I dont have any 2.0 hubs from anyone really reputable but all three of the 3.0/3.2 hubs I've tried have worked with the 8bitdo v2, I haven't opened them up to see what's in there though.