Ryzee119/ogx360 #104: "Waiting for device" problem

Miggas, 2023-03-25:
Hi, I've made a custom board and after assembling it and testing, no controllers where detected, I've already triple checked all connections to make sure everything is connected to the right pin and the solder joints are good, but I'm still getting this issue,
it stops on "Waiting for device" and stays there even with other controllers.
[image]

Miggas, 2023-03-26:
[custom pcb.zip](https://github.com/Ryzee119/ogx360/files/11070542/custom.pcb.zip)
Here's the files anyway

Ryzee119, 2023-03-26:
I can't really provide support for custom boards but check the USB port is getting 5V and the D+/- are the right way around

closed 2023-04-07 by Ryzee119

Miggas, 2023-04-10:
Took a while, but I've found the problem, the voltage regulator ic was faulty and was supplying 4.9v instead of 3.3v, I replaced it and now it works fine. Even though the issue is closed, I'll leave this info here to help someone in case of the same problem.