Ryzee119/ogx360 #106: Additonal buttons on the BOM?

qwertychouskie, 2023-04-07:
The BOM has 5 of the [EG5350CT-ND](https://www.digikey.com/en/products/detail/e-switch/TL3305AF160QG/5816181) buttons, however, as far as I can tell, the board only has space for one button.  Is this a mistake or am I missing something?

Ryzee119, 2023-04-07:
The BOM just rounded up to closest group of 5 or 10 units depending of price of the item

qwertychouskie, 2023-04-07:
The number 5 directly comes from https://github.com/Ryzee119/ogx360/blob/master/.kitspace/bom.csv despite minimum order quantity being 1.  Now that I'm looking closer at it, many other components have higher numbers too (e.g. 5 clock crystals instead of one).

Ryzee119, 2023-04-07:
Right it's just rounded up to closest 5 or 10 depending on price.

PR #107 (2023-04-18): Make component quantities in BOM correct
Updated quantities are based on the labels listed (e.g. `"C2,C3"` would be a quantity of 2).

Also adds a note to the relevant readme to suggest ordering a couple extra of the tiny components if hand-assembling/soldering.

Fixes #106
closed 2023-04-18 by Ryzee119