Ryzee119/ogx360 #108: Upgrade Master to ATMega64

XeniumMods, 2023-04-17:
Would it be possible to upgrade the master from an ATMega32 to ATMega64?
Or even an ATMega128 as they share the same pinout as the ATMega64...

I figured just the master could be upgraded where the salves remain the same.
This should open up the possibilities of being able to add additional features.

I would be happy to contribute anyway I can... For example, if you were to do a
schematic, I could make a proto PCB design and assemble a handful for testing.

Either way it would be great to discuss this further with you if your interested.
Is there some way we can communicate directly?

Cheers, Nem

Ryzee119, 2023-04-17:
https://www.sparkfun.com/products/18288

Does this have the same pinout?

XeniumMods, 2023-04-18:
[image]

The ATmega32u4 (32kb) is 44pin whereas the ATmega64 (64kb) has 64pins.
Note the ATmega128 (128kb) is also 64pin with the same pinout as ATmega64.

Even then the pinouts are still fairly similar as shown above...
I marked out the IO in Yellow and everything else in Blue.

The most obvious thing that the ATmega64/128 are missing is the USB Host controller.
I put together 3 block diagrams to help show 3 different hardware solutions.

XeniumMods, 2023-04-18:
[image]

Probably the most easiest solution would be to add an addition ATmega32u4 slave for Player 1.
Or from the perspective of the current hardware we are adding the ATmega64 to the design.

XeniumMods, 2023-04-18:
[image]

An alternative solution would be to use a dual USB host controller that will serve as both the
Input USB Port and Output USB Port for Player 1. This solution maybe to difficult to implement.

XeniumMods, 2023-04-18:
[image]

I believe Diagram 3 would be the most efficient and cost effective use of hardware.
Instead of using an ATmega32u4 slave for every port, if we could just the use 1 that
could drive a 4 port USB Hub controller. This would definitely make the hardware
far more simpler and compact thus reducing the number of components and costs.

XeniumMods, 2023-04-18:
[image]

**Benefits of using a more capable MCU...**

_Firstly I'd like to discuss what's already supported by the ATmega32:_

**Xbox360 Wireless Receiver:**

The device acts in a similar manner to an Xbox 360, allowing up to 4 controllers and 4 headsets
at a time to be connected to the receiver.

**8BitDo v2 Bluetooth Receiver:**

Can only support one controller at a time so 4 receivers would be required for all 4 players.
However it can connect to a wide array of Bluetooth controllers including:

- XboxONE (Bluetooth versions only),
- Xbox Series X/S,
- Playstation 3, 4 & 5
- Nintendo Switch and Joy-Con

_Secondly what could be supported in the future using the ATmega64:_

**Xbox "Series X/S" Wireless Receiver:**

This device can connect with Xbox controllers, headsets, and similar accessories via Bluetooth.
Being able to connect to 8 controllers at once, supporting both Bluetooth Classic and Low Energy.
From what I can tell it's a cutdown version of the XboxONE wireless receiver.

**Xbox "ONE" Wireless Receiver:**

Is a larger/older version of the Series X/S wireless receiver however this device can connect with
Xbox controllers, headsets, and similar accessories via the proprietary Xbox Wireless protocol rather
than just Bluetooth. From what I've tested also supports both Bluetooth Classic and Low Energy.
Having a 2.4Ghz transceiver this device supports non-Bluetooth XboxONE controllers.

**8Bitdo Xbox Remote Control:**

This device is a remote control for XboxONE and Series X/S consoles. There are two versions,
a full size black version with additional buttons and a white version which is more cutdown.
It would be amazing to see added support for this remote control in replacement of the
original xbox remote control. 

Perhaps adding IO to the ATmega64 to support Powering OFF the console and Ejecting the
DVD tray. And a few more to trigger actions with a ARGB controller as for example.
Buttons 0 through to 9 on the full size remote control could be used in this way...

_I hope this hasn't been to overwhelming, I really needed to get this all out of my head ;)
Will be interested in what you think..._

Cheers, Nem

Ryzee119, 2023-04-18:
> Instead of using an ATmega32u4 slave for every port, if we could just the use 1 that
could drive a 4 port USB Hub controller.

This is not how USB hubs work. we need 4 independent usb device controllers.

My point with linking the RP2040 dev board is that it looks at a quick glance to have the exact pinout as the Arduino pro micro so would be a direct swapout for the master module. It wouldnt be alot of work to recompile to RP2040. Main difference being a USB C port instead of USB Micro

XeniumMods, 2023-04-20:
> It wouldnt be alot of work to recompile to RP2040. Main difference being a USB C port instead of USB Micro

I agree USB type C port would be more durable and user friendly but programming a new MCU for it
seems like a lot of work, personally I would implement type C via hardware as I generally don't rely on
development boards.

> This is not how USB hubs work. we need 4 independent usb device controllers.

Oh right.... I don't have much experience working with USB controllers / hubs. Do you have any other
ideas for a minimalistic design that can support 4 controllers?

* Is it possible to drive two or more MCU's from a single oscillator?

Or is Diagram 1 where we could drop-in an ATmega64 into the current design appeal to you more?
I wouldn't think it would be to difficult in migrating the current Atmega32 master firmware to
ATmega64 and then adding an extra ATmega32 slave to the firmware for Player 1.

XeniumMods, 2023-04-26:
[image]

I would be happy to supply you a couple of each of the xbox wireless receivers.
They are the aftermarket versions which seem to be available everywhere.
I could also supply you with a 8bitdo xbox remote controller.
The image is of the one I've been using with my series x for the past year.
But I can get you a new remote if you like?

Ryzee119, 2023-05-02:
The atmega 64 or 128 doesnt support USB either actually so these would be no good. Side note , I grabbed one of the pro micro rp2040s and it fits and Ive got it compiling and talking with the USB controllers already.

The USB device size (Xbox facing) API is a bit different so thats not working yet but not hard. It doesnt make loads of sense to use this on the ogx360 and its entirely overkill (Flash used gots from 100% down to like 0.01% haha) and probably would be much better of with its own much simpler design as this supports USB Host interrnally so we dont need a max341 chip.

[image]

XeniumMods, 2023-05-03:
Interesting...

Alternatively I would have suggested "STM32" but seeing that I can buy standalone "RP2040TR7" MCU from element14 it would still be possible to do custom PCB design, actually the price for a full reel is very appealing.

https://au.element14.com/raspberry-pi/rp2040tr7/mcu-32bit-133mhz-qfn-56/dp/3766079

Quote: "Support for up to 16MB of off-chip Flash memory via dedicated QSPI bus"

Think that Windbond W25Q128JVPM has 16Mb of flash memory which may explain why your at 0.01%
Could only find datasheet for "W25Q128JVPIM", I'm assuming its the same IC...

https://www.digikey.com/en/products/detail/winbond-electronics/W25Q128JVPIM-TR/6819720

**Side Note:** Would you be interested in some private work?

Just finished my dual 80mm fan caddy to replace the original HDD caddy.
Could definitely use a PWM fan controller based on M/B and CPU temps.
Your "spi2par" would make a great starting point to acquire temps via i2c.

Miggas, 2023-05-10:
I was reading all the comments and would like to suggest the ATSAMD21, it can do both USB Device and USB Host, is really cheap and can be programmed like an arduino (Arduino IDE, PlatformIO)

XeniumMods, 2023-05-12:
> ATSAMD21... is really cheap.

For the sake of research I compared prices.

**Development Boards:**

RP2040 = $10.95aud
[https://www.sparkfun.com/products/18288](url)

SAMD21 = $22.50aud
[https://www.sparkfun.com/products/13664](url)

**Micro Controllers:**

RP2040TR7 = $1.21aud each @ 500pcs (min)
[https://au.element14.com/raspberry-pi/rp2040tr7/mcu-32bit-133mhz-qfn-56/dp/3766079](url)

ATSAMD21G18-AU = $6.08aud each @ 100pcs (max)
[https://au.mouser.com/ProductDetail/Microchip-Technology/ATSAMD21G18A-AU?qs=KLFHFgXTQiDh1vL0VLU76Q%3D%3D](url)

**Conclusion:**

Considering each 4-channel OGX360 requires 4x dev. boards or MCU's the price would add up quickly.
Then if you were to make 100x OGX360's then the price different is completely out of proportion.
Comparing dev. boards the price difference for 100x OGX360's would be $4620aud or $46.20aud per unit.
This is worse case scenario as I'm sure if one looked around you could get better bulk rates on dev. boards.

@Ryzee119 - I'll be ordering some PCB's in the next week or two if you need anything?

Ryzee119, 2023-05-12:
If you want cheap use a ch559 for the host controller and 4x ch552 for the devices. They are like 50c each or less in qty

XeniumMods, 2023-05-14:
Interesting...

So I made an enquiry over at Elecrow since if I were to get some PCBs
made up that's where I would likely to go. 

RP2040 = $0.92/unit (@ 500pcs)
CH559L = $1.15/unit (@ 100pcs)
CH552T  = $0.39/unit (@ 100pcs)

The RP2040 and CH559L are comparable, makes me beg the question
if it's still possible still use RP2040 as host controller instead of CH599L?

I know I haven't factored in the flash memory still required for RP2040
or that there's a minimum order of 500pcs at this stage.