Ryzee119/ogx360 #110: Steel Battalion Controler Emulation support for the Game "Steel Battalion - Line of Contact"?

xfortknoxX, 2023-05-27:
Thanks for your recent help here:
[https://github.com/Ryzee119/ogx360/issues/105](url)

I finally got a QWERTY Layout chatpad and the Singleplayer Game "Steel Battalion" works fine with it. 👍 

Now I wonder wether the Steel Battalion controller emulation is compatible in its current state with the Game "Steel Battalion - Line of Contact"?

Cause after I tried it, it seems to recognize the controller properly at first (I can press successfully press the "main weapon button" and get to the main menu of the game)
But after choosing
"Xbox Live"
or
"System Link"
I always get this error and cannot continue:

[image]

Switching back to the duke and then back to the controller Emulation didn't solve the issue for me.

You have the same issue? Or is this game simply not yet compatible?

Thank you in advance Ryzee119

Info:
I'm using an ogx360 with the latest FW

Ryzee119, 2023-05-27:
I have a feeling that games needs a normal controller in one port. Can you try a normal controller on port 1 and ohx360 on the other port?

xfortknoxX, 2023-05-27:
Thank you for the fast response.
Good idea. I'll try that out later and give you my feedback then.
Greetings

xfortknoxX, 2023-05-27:
You were right.
I connected a common wireless controller to ogx360 port 1 and connected the controller with the chatpad to the second port.
Then it worked flawlessly.

Strange behavior. Are there other games in existance with the same requirements?

Thank you for everything ☺️

Ryzee119, 2023-05-27:
Not that I'm aware of. That game just expect a controller to navigate menus and you wouldn't normally unplug it and connect your steel battalion controller back and forward

closed 2023-05-27 by Ryzee119