Ryzee119/ogx360 #111: Add support for Xbox 360 microphones

JaviJJ117, 2023-09-25:
Hello, it would be very useful to add compatibility with Xbox 360 microphones for online gaming.
It would be great if the standard headphones that connect with a jack to the controller and the wireless headphones that connect with Bluetooth to the Xbox 360.
I'm sure it can be done.

Ryzee119, 2023-09-25:
Max3421 host controller doesn't support isochronous USB transfers so it is not possible with this hardware unfortunately.

The atmega32u4 is also at capacity with its current feature set.

closed 2023-09-25 by Ryzee119

JaviJJ117, 2023-09-26:
That is quite discouraging, is it possible that in the future you will release a second version of the ogx360 with different hardware capable of supporting isochronous USB transfers?